raspberrypi/firmware #871: net boot isn't respecting option 66

roysjosh, 2017-09-09:
My router at 192.168.10.1 is serving DHCP but not TFTP.  I am attempting to redirect TFTP to 192.168.10.100.  I have set server-ip-address (siaddr), tftp-server option 66 as well as the PXE server IP address list under option 43.  My pi3 is booting from a sdcard with a bootcode.bin from yesterday.  I have a suspicion it is due to dnsmasq intentionally including a trailing `\0` on the tftp-server.

dnsmasq.conf snippet:
```
pxe-service=0,"Raspberry Pi Boot   ",1,192.168.10.100
dhcp-option=option:server-ip-address,192.168.10.100
dhcp-option=option:tftp-server,"192.168.10.100"
```

tcpdump:
```
00:44:52.298732 b8:27:eb:0f:69:01 > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 364: (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 350)
    0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from b8:27:eb:0f:69:01, length 322, xid 0x26f30339, Flags [none]
	  Client-Ethernet-Address b8:27:eb:0f:69:01
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Discover
	    Parameter-Request Option 55, length 14: 
	      Vendor-Option, Vendor-Class, BF, Option 128
	      Option 129, Option 130, Option 131, Option 132
	      Option 133, Option 134, Option 135, TFTP
	      Subnet-Mask, Default-Gateway
	    ARCH Option 93, length 2: 0
	    NDI Option 94, length 3: 1.2.1
	    GUID Option 97, length 17: 0.1.105.15.72.1.105.15.72.1.105.15.72.1.105.15.72
	    Vendor-Class Option 60, length 32: "PXEClient:Arch:00000:UNDI:002001"
00:44:55.303716 58:6d:8f:ed:aa:18 > b8:27:eb:0f:69:01, ethertype IPv4 (0x0800), length 421: (tos 0xc0, ttl 64, id 19584, offset 0, flags [none], proto UDP (17), length 407)
    192.168.10.1.67 > 192.168.10.237.68: BOOTP/DHCP, Reply, length 379, xid 0x26f30339, Flags [none]
	  Your-IP 192.168.10.237
	  Server-IP 192.168.10.100
	  Client-Ethernet-Address b8:27:eb:0f:69:01
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Offer
	    Server-ID Option 54, length 4: 192.168.10.1
	    Lease-Time Option 51, length 4: 3600
	    RN Option 58, length 4: 1800
	    RB Option 59, length 4: 3150
	    Subnet-Mask Option 1, length 4: 255.255.255.0
	    BR Option 28, length 4: 192.168.10.255
	    Default-Gateway Option 3, length 4: 192.168.10.1
	    TFTP Option 66, length 15: "192.168.10.100^@"
	    Vendor-Class Option 60, length 9: "PXEClient"
	    GUID Option 97, length 17: 0.1.105.15.72.1.105.15.72.1.105.15.72.1.105.15.72
	    Vendor-Option Option 43, length 44: 6.1.3.10.4.0.80.88.69.8.7.0.1.1.192.168.10.100.9.23.0.1.20.82.97.115.112.98.101.114.114.121.32.80.105.32.66.111.111.116.32.32.32.255
00:44:55.304226 b8:27:eb:0f:69:01 > 58:6d:8f:ed:aa:18, ethertype IPv4 (0x0800), length 77: (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 63)
    192.168.10.237.49153 > 192.168.10.1.69:  35 RRQ "480f6901/start.elf" octet tsize 0
00:44:56.692428 b8:27:eb:0f:69:01 > 58:6d:8f:ed:aa:18, ethertype IPv4 (0x0800), length 71: (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 57)
    192.168.10.237.49154 > 192.168.10.1.69:  29 RRQ "autoboot.txt" octet tsize 0
00:44:58.788591 b8:27:eb:0f:69:01 > 58:6d:8f:ed:aa:18, ethertype IPv4 (0x0800), length 69: (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 55)
    192.168.10.237.49155 > 192.168.10.1.69:  27 RRQ "config.txt" octet tsize 0
00:45:00.893064 b8:27:eb:0f:69:01 > 58:6d:8f:ed:aa:18, ethertype IPv4 (0x0800), length 71: (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 57)
    192.168.10.237.49156 > 192.168.10.1.69:  29 RRQ "recovery.elf" octet tsize 0
00:45:02.993983 b8:27:eb:0f:69:01 > 58:6d:8f:ed:aa:18, ethertype IPv4 (0x0800), length 68: (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 54)
    192.168.10.237.49157 > 192.168.10.1.69:  26 RRQ "start.elf" octet tsize 0
00:45:05.096605 b8:27:eb:0f:69:01 > 58:6d:8f:ed:aa:18, ethertype IPv4 (0x0800), length 68: (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 54)
    192.168.10.237.49158 > 192.168.10.1.69:  26 RRQ "fixup.dat" octet tsize 0
```

ghollingworth, 2017-09-09:
Get rid of the "raspberry pi boot" from the reply, this means that the DHCP
server is also the tftp server and it'll process both overwriting the
option 43

roysjosh, 2017-09-09:
Ah, that did the trick.  Thanks!

closed 2017-09-09 by roysjosh