raspberrypi/firmware #872: raspberry pi 0 enc28j60 driver start but eth0 is not enable

mrclemds, 2017-09-11:
I followed many tutorial on how to add a ethernet port to the raspberry pi 0 with the enc28j60 module i've buy but when i plug a cable no led light up until i unplug/replug vvc or ground pin from the module and then it's detected by the other device but the raspberry pi itself does nothing.
Sorry for my english ...

ifconfig :
[image]

lsmod :
[image]

dmesg :
[image]

fedorovvl, 2017-09-12:
you tried command "ifconfig eth0 up"?

mrclemds, 2017-09-13:
Nop i was thinking "ifup eth0" is the same.
what's the difference with "ifup eth" ?
I find that's the /etc/network/interfaces file who cause this because it was empty.
Thank you for the help.

closed 2017-09-14 by mrclemds