raspberrypi/firmware #873: sha256sum changed for rpi-firmware-....tar.gz downloaded from github

user1048576, 2017-09-12:
Hi,

This may not be the correct place to ask this question, but I'll give it a try anyway to see if anyone else has had the same issue.

I regularly download rpi-firmware form this repository on github as part of a build process. In most cases I keep the downloaded tar.gz -file, but sometimes a clean it away and download it again.

I noticed that today, 20170912, the sha256sum for the downloaded tar.gz changed, compared to one tar.gz downloaded 20170829.

As far as I can see, github does not provide any digest to verify such downloads, and the sha256sum is locally generated, at a previous occasion. But it has been the same for several downloads in the past. So somehow the content of rpi-firmware-b9dbf8cf1a4609d16304c3f2740e97d5057d9a1a.tar.gz downloaded from github has changed, which is a bit puzzling.

Command I use to download:
wget https://github.com/raspberrypi/firmware/archive/b9dbf8cf1a4609d16304c3f2740e97d5057d9a1a/rpi-firmware-b9dbf8cf1a4609d16304c3f2740e97d5057d9a1a.tar.gz

the sha256sum's I get:
$ sha256sum rpi-firmware_20170829/rpi-firmware-b9dbf8cf1a4609d16304c3f2740e97d5057d9a1a.tar.gz rpi-firmware_20170912/rpi-firmware-b9dbf8cf1a4609d16304c3f2740e97d5057d9a1a.tar.gz 
ce744f55bf9105f20f6111c7810055414514564b8a73ba536bb31c532f369002  rpi-firmware_20170829/rpi-firmware-b9dbf8cf1a4609d16304c3f2740e97d5057d9a1a.tar.gz
5facbbba4cc897c39784a9775c14a6ac30230a3c14ef47e78f22eb219dd3fe7b  rpi-firmware_20170912/rpi-firmware-b9dbf8cf1a4609d16304c3f2740e97d5057d9a1a.tar.gz
$

Has anyone else seen the same issue ? Maybe it is not an issue, couldn't find anywhere that github actually provide any digest for such downloads, and since those tar.gz -files are possibly created "on-the-fly" when they are requested from the repository, this might be nothing to bother about. I _do not_ suspect that the contents of the repository has changed.

br

popcornmix, 2017-09-12:
Do you have both tar.gz files?
Can you compare the contents after extracting? Are any files different?
It is possible github has updated to newer version of the archiving tool (I suspect the archives are created on-the-fly) and there are differences in archive header or compression used.

user1048576, 2017-09-12:
Yes, I have both. I unpacked them, and it seems the sha256sum for each individual file in the tar.gz is the same. So your suggestion about github having a new archiving tool seems plausible.
It is a bit annoying however, but I understand it is nothing that the maintainers of this repository has anything to do with.
(I'm using buildroot, and I have a locally generated sha256sum that buildroot compares against after having downloaded a tar.gz, and now the build fails. But I guess I just have to change the locally generated sha256sum, or skip the check altogether)

popcornmix, 2017-09-12:
I guess belt-and-braces would be to check sha256sum of archive and if it fails extract and check contents. If that matches then update the archive sha256sum.
Could all be automated and the slower extracted check will only be triggered very occasionally.

user1048576, 2017-09-12:
Yes, that would be a safe approach. I think I'll leave it as it is for this time, just update the sha256sum in the make-file. Hopefully it is nothing that happens too often, and maybe it is not worth the effort to make an automated check for this. Thanks for your attention

closed 2017-09-12 by user1048576

MilhouseVH, 2017-09-14:
LibreELEC recently began using `sha256sum` as the checksum for downloaded package tarballs and we've noticed 122 out of 321 github tarball checksums have changed in the last few days.

I emailed github support asking them to confirm this is an intentional change (before we go ahead and update all the changed checksums), and assuming it is suggested a blog post about the change might be in order. This is their reply:
```
Hi Neil,

Thanks for reaching out about this! We recently applied a Git bug-fix which changes
the tarball representation of filenames longer than 100 characters, and that's the
immediate cause of hashes changing. You can see the upstream commit that we applied
here:

https://github.com/git/git/commit/22f0dcd9634a818a0c83f23ea1a48f2d620c0546

As a general rule we encourage people not to rely on byte stability of tarballs.
That being said, it's definitely something we're discussing internally, because we
know it's something that would be convenient for people to be able to rely on.
There's some public discussion with comments from our engineers in these issues, in
case you'd like some background:

https://github.com/libgit2/libgit2/issues/4343#issuecomment-328631745
https://github.com/Homebrew/homebrew-core/issues/18044#issuecomment-329313468

I'll definitely pass your feedback along about the blog post. In the meantime, I
hope that helps explain things, but let us know if you have any other questions!

All the best,
Rachel
GitHub Support
```