raspberrypi/firmware #876: Framebuffer size not matching HDMI resolution

TorbenSchlsng, 2017-09-18:
Hello,

I am using the hdmi_timings command in config.txt to create custom video timings. Unfortunately, there seems to be a problem with the framebuffer size. If I set
`hdmi_timings=1100 1 350 40 40 790 1 21 2 4 0 0 0 25 0 31250250 1`
the framebuffer size reported via `ioctl()` is 3488640 but should be 3476000 for 32bits per pixel. The x- and y-resolution reported via `ioctl()` are fine (1100/790). `tvservice -s` and `fbset -s` are showing the same value. Of course this results in some strange video output on the display...

popcornmix, 2017-09-18:
It sounds like you are not accounting for the stride.
The GPU has an alignment requirement of16 pixel, so the stride will be 1104 pixels which accounts for the size reported.
You need to use the stride rather than the width when determining the address of the line of pixels below the current one.

TorbenSchlsng, 2017-09-18:
Thank you for the fast reply! I didn´t know about this... Is there a documentation where I can find this information?

popcornmix, 2017-09-18:
Try https://www.kernel.org/doc/Documentation/fb/api.txt
They call the stride parameter ``line_length``.

TorbenSchlsng, 2017-09-18:
Thanks! I read this sometime ago, but didn`t notice it. Thank you for the fast support!

closed 2017-09-18 by TorbenSchlsng