raspberrypi/firmware #877: Export cached RSTS value via MBOX interface

steelman, 2017-09-20:
u-boot will greatly benefit from knowledge of RSTS value because it can take different route to boot a device depending on reboot reason and reboot partition (although that is already "consumed"). u-boot has implemented MBOX interface. Unfortunately RSTS isn't exported via this interface.

Please add a tag to read RSTS via MBOX interface.

popcornmix, 2017-09-21:
Sounds reasonable. I'll add this when I get a chance.

steelman, 2017-09-21:
Thank you in advance.

popcornmix, 2017-10-03:
Added this. Will be in next rpi-update firmware
```
vcgencmd get_rsts
get_rsts=2
/opt/vc/bin/vcmailbox 0x00040011 8 8 0 0
0x00000020 0x80000000 0x00040011 0x00000008 0x80000004 0x00000002 0x00000000 0x00000000
```
The sixth word output should match the ``vcgencmd get_rsts`` value.

popcornmix, 2017-10-06:
This is in latest rpi-update firmware

steelman, 2017-10-06:
Thanks. I will check on Monday. BTW, what are the other arguments ("8 8 0 0") and results?

popcornmix, 2017-10-08:
There is only one useful work of result. The rest is the protocol for the mailbox interface (e.g. lengths of messages/results). https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface

JamesH65, 2019-01-10:
Closing this issue as questions answered/issue resolved.

closed 2019-01-10 by JamesH65