raspberrypi/firmware #878: Fallback to the default partition if RSTS points to a nonexistent one

steelman, 2017-09-21:
I'd like to propose a clarification of the reboot protocol that involves passing a partition number in the RSTS register. As far as I can tell after a day of duckduckgoing, it is not throughly described in any official document. So I guess my proposal, if implemented, won't break stuff.

At the moment, bootcode.bin (AFAIK) reads the RSTS and loads (or is it start.elf) kernel (?) not from the default partition but rather from one pointed by some bits (0, 2, 4, 6, 8, 10) in the register. If anything goes wrong the code stops and blinks the ACT LED.

I propose, that the firmware (bootcode.bin?) should fall back to the default partition (1) if the value in  the RSTS points beyond the last existing partition. After #877 is implemented, it is going to be possible to use the RSTS register not only to control the firmware but also bootloaders like u-boot or any other code able to access the value. It will be also possible, if OS/bootloader vendor chooses, distinguish watchdog reset from reboot,  which AFAIK is not possible at the moment.

For example. There are 10 partitions on an SD card. If the value in the RSTS stored before a reboot is less than or equal 10, the firmware works as it does today. If the number is larger than 10 and less than 63, the kernel is loaded from the default partition. However the number is available via get_rsts and MBOX (see #877) for the software to read.

popcornmix, 2017-10-03:
The bootrom will read from the first partition and load bootcode.bin from there.
bootcode.bin will use the bits in RSTS register to read from the specified partition.
If this partition does exist then it loads start.elf from this partition.
If this partition does not exist then it falls back to loading start.elf from first partition.
You should be able to test this on raspbian:
```
echo 62 > /sys/module/bcm270?/parameters/reboot_part
reboot
```
This should reboot (from partition 1 assuming partition 62 does not exist).
You can then use ``vcgencmd get_rsts`` to distinguish this reboot from a normal reboot.

popcornmix, 2017-10-03:
Actually that has changed on 4.9. See https://github.com/raspberrypi/linux/issues/2095 for new way of specifying reboot partition.

steelman, 2017-10-10:
Allow me to make sure, we are on the same page. As of today the "fallback" feature isn't yet implemented, is it?

When issue the command `reboot 62` I get the following messages on the console
```
[  570.304295] reboot: Restarting system with command '62'

Raspberry Pi Bootcode
Read File: config.txt, 1868
Read File: start.elf, 2827300 (bytes)
Read File: fixup.dat, 6579 (bytes)
```
and the green LED keeps blinking repeatedly seven times in a row.

popcornmix, 2017-10-10:
7 flashes is "kernel not found" which suggests bootcode.bin and start.elf were loaded successfully.
Do you have a kernel?

steelman, 2017-10-10:
I believe, I do. (I don't have the device at this very moment).

a) I've got only one FAT partition with firmware files, u-boot and kernel.

b) I've got `kernel=u-boot.bin` in `config.txt`.

c) When I don't set RSTS explicitly (powe-on, reboot without an argument), u-boot is loaded properly and does its job of loading and starting the kernel.

-- 
Łukasz Stelmach z podróży

steelman, 2017-10-17:
Ping?

steelman, 2017-12-28:
Ping? Anything?

JamesH65, 2019-01-10:
Did you get anywhere with this?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

steelman, 2019-01-10:
It's funny that I checked this issue in the morning. I've got new binaries on a card but did some other work today. I will see tomorrow. Thanks.

steelman, 2019-01-11:
Apparently something has changed, but not the way I had requested above. Now when I reboot with an argument, the board reboots, reads boot files from partition 1 (the only FAT partition holding bootcode.bin config.txt et al.), starts the configured kernel (u-boot in my case, to be precise) and the RSTS is set to 0x20.

```
root:~> /sbin/reboot 4
Rebooting with argument '4'.
[  423.940666] reboot: Restarting system with command '4'

Raspberry Pi Bootcode
Read File: config.txt, 2127
Read File: start_x.elf, 3955844 (bytes)
Read File: fixup_x.dat, 9875 (bytes)
MESS:00:00:01.056585:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.061622:0: brfs: File read: 2127 bytes
MESS:00:00:01.075481:0: HDMI:EDID error reading EDID block 0 attempt 0
[…]
U-Boot RPI3 SPL 2017.03-gf6f75b5bb-TIZEN.org (Jan 04 2019 - 06:13:14)
U-Boot 2017.03-gf6f75b5bb-TIZEN.org (Jan 04 2019 - 06:13:14 +0000)
DRAM:  896 MiB
RPI 3 Model B (0xa02082)
[…]
U-Boot> echo $reg_rsts
0x00000020
````

The same happens if the reboot argument number is larger than the number of partitions on the sdcard.

JamesH65, 2019-11-05:
@popcornmix @timg236 Any thoughts, or close?

popcornmix, 2019-11-06:
As far as I'm aware this is working as intended. Using a non-existing partition number will fall back to booting from the first partition and the partition number can be read back. e.g.
```
pi@pi4d:~ $ vcgencmd get_rsts
get_rsts=20
pi@pi4d:~ $ sudo reboot 10
...
pi@pi4d:~ $ vcgencmd get_rsts
get_rsts=64
pi@pi4d:~ $ sudo reboot 11
...
pi@pi4d:~ $ vcgencmd get_rsts
get_rsts=65
```

closed 2025-02-14 by popcornmix