raspberrypi/firmware #879: /opt/vc/include/GLES2 headers conflicting with libgles2-mesa-dev

cetex, 2017-09-22:
I'm trying to build QT Wayland and it seems like the header definitions provided in:
/opt/vc/include/GLES2/gl2.h are conflicting with the headers provided by libgles2-mesa-dev in /usr/include/GLES3/gl3.h

When i check khronos reference headers they have added an ifndef around the whole block of defines in gl2.h: https://github.com/KhronosGroup/OpenGL-Registry/blob/master/api/GLES2/gl2.h#L59

I think this should be added in videocore to make builds work.

cetex, 2017-09-22:
My understanding of this is quite shallow, but i've found others dealing with the same problem elsewhere:
https://www.khronos.org/bugzilla/show_bug.cgi?id=761

http://code.qt.io/cgit/qt/qtwayland.git/commit/?id=d5e7965a87d81e2d7157c785403b0aba681de62a (the workaround in QT does not seem work)

JamesH65, 2019-01-10:
Current version of gl2.h appears to have a similar protection define,  so not sure tha tis the problem. Sounds like this is aproblem not only in our tree though.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-11-05 by JamesH65