raspberrypi/firmware #881: Cannot take "high detail" snapshot of current screen without garbled images

ephestione, 2017-09-26:
I am using a setup where a combination of `screen` and `omxplayer` (as per [this article](https://community.ubnt.com/t5/UniFi-Video/Tutorial-RTSP-Raspberry-Pi-B-Viewer-6-Cam-4-Cam/td-p/1536448)) are showing a 2x2 grid of RTSP videos from 3 cameras, so the bottom right rectangle is dark. My cameras are Xiaomi XiaoYi Ants, and they output a 720p video which I can choose to be low or high bandwith (higher or lower compression).

They display fine on the FullHD monitor the raspi is attached to (wifi allowing); if I use [screenshot](https://github.com/info-beamer/tools/tree/master/screenshot) to grab the screen when all of them are on low quality, or just one is on high quality, the captured image is fine, but when I have 2 or all 3 streams in high quality, artefacts appear that do not show on screen, like the attached image
[image]

(the date on top left is added from the original JPEG saved by `screenshot` by a PHP code with GD module, and the big white rectangles with red outline are crops I made for privacy reasons)

I tried with [raspi2png](https://github.com/AndrewFromMelbourne/raspi2png) as well, and I get the exact same artefacts.

The issue on screenshot's GIT is here: https://github.com/info-beamer/tools/issues/5

On the thread for the RTSP monitor tutorial (linked at the beginning) there are people reporting pretty much a similar thing, where they get a blank video frame as soon as they try to show 6 streams together: up to the fifth they show correctly.

popcornmix, 2017-09-27:
Does adding "dispmanx_offline=1" to config.txt (and rebooting) make any diffference?

6by9, 2017-09-27:
From your linked infobeamer issue, dispmanx is being asked to compose
```
display:2 format:YUV_UV transform:0 layer:0 src:0,0,640,360 dst:960,0,960,540 cost:683 lbm:10240
display:2 format:YUV_UV transform:0 layer:0 src:0,0,640,360 dst:0,540,960,540 cost:683 lbm:10240
display:2 format:YUV_UV transform:0 layer:0 src:0,0,1280,720 dst:0,0,960,540 cost:1145 lbm:15360
```
So it appears your low quality streams are 640x360, and the high quality ones are 1280x720. (That contradicts your statement above that they output a 720p video - they appear not to always). That explains the increased cost and lbm values for the streams.

There is a limit to the buffering available in the hardware, and that is increased with particularly vertical downscaling of images. That's the most likely issue. Not sure how that buffer is split between online (HDMI) and offline composition.

dividuum, 2017-09-27:
I never thought about this, but I guess since usually the HVS composes directly to HDMI, the snapshot done with `vc_dispmanx_snapshot` also has to trigger the composer and renders into some kind of buffer? Is that correct? Any idea why this artefact only appear while taking a snapshot?

If it's due to downscaling, placing/scaling the videos in a way that the video size matches the output size should help? So `omxplayer --win '0,0,640,360' rtsp://...` for the first 640x360 video, `omxplayer --win '640,0,1280,360' rtsp://...` for the second video and so on. Preferably without them overlapping. @ephestione, can you try that?

6by9, 2017-09-27:
> Any idea why this artefact only appear while taking a snapshot?

At a guess because it is being naive and splitting the available buffering between the online HDMI and offline snapshot composition. popcornmix will know more.
It's possible that the one HDMI frame generated at the same moment as the snapshot has the same artifacts, but when it's running at 60fps you're not really going to see that.

The cost and buffer allocation is something I'm not that familiar with, but simplistically to scale the 720p images into the apparent 960,540 windows will always require 2 lines of input per line of output. Even if the windows are reduced to 640x360 then the same is also true.
Displaying 640x360 into those windows will typically mean that two (shorter) lines are required to interpolate between them.
Overlapping hurts as more data has to be retrieved in order to compose the same amount of output, but I'm not sure that will have much impact in this case.

The suggestion of "dispmanx_offline=1" is still valid as it shifts the HDMI composition to being into a buffer instead of realtime. It may allow the HVS to assign the buffering entirely to one task at a time.

ephestione, 2017-09-27:
First of all, thank you all for the replies.
In the thread at `screenshot`'s GIT issues, I hypothesized that it was indepoed a buffer problem where the direct HDMI output took precedence and displayed fine, while the snapshot routine "piggybacked" and ended up filling the buffer, hence the garbled screen.
Well anyway popcornmix nailed it right off the bat, or at least I suppose! I am ssh'ing into the raspi3 now, and hae no means to check wether the HDMI display is showing ok (said display is even turned off now by a hardware switch on the wall), but sure enough after having added
`dispmanx_offline=1`
on config.txt and rebooting, and enabling the high quality stream (which is indeed a lower resolution rather than same resolution and higher compression, I'm sorry for that!) on a second camera, I now get perfectly fine screenshots, no artefacts whatever.

Regarding @dividuum 's suggestion with the window size, I was going to try it next if popcornmix's didn't work, yet it was a suboptimal path in my case because then I would have had to make do with smaller video frames in my 22" monitor, and well, they are never big enough, I wouldn't like for them to get even smaller :)

I can just keep `dispmanx_offline=1` (have yet to see the effects on the real life HDMI image display, tomorrow I will be able to give feedback), but maybe there's something yet that can be done so that directive is not needed?

ephestione, 2017-09-27:
No, sorry, I was fooled.
I don't know how, why, where or what, but just like the first screenshots I loaded from the small webserver I had hacked together to serve them where fine, at least for the first few minutes, now all of them appear exactly as they did before.
Every single one, garbled with the greenish noise.

ephestione, 2017-09-27:
So, for science, I also went with @dividuum theory, and keeping `dispmanx_offline=1` I resized the --win sizes as:
0 0 1280 720
1280 0 1920 360
640 360 1920 1080
according to each frame's original resolution (I had to overlap first and third since they cannot fit otherwise)
and, **for now**, everything shows fine on my remote side (albeit the overlapping detail, still no idea about the hdmi image on the monitor).
If everything is still fine later, I will try commenting the dispmanx_offline=1 in config.txt and reboot to check if the solution still works.
If this was the solution, I would still go back to having just one high resolution frame and two low resolution, because otherwise the system is not really useable. Still, if this was enough to get rid of the artefacts, it would be a pointer to what to change, right?

ephestione, 2017-09-27:
I think I can be definitive by now: with dispmanx_offline=1 commented out, @dividuum 's method still gives back working screenshots, and it's been like this for 20 minutes, while as soon as I switch from 2x2 grid and same streams, screenshots are all garbled, and they get so immediately after the switch.
I **still** have no idea how the HDMI output looks like though, as I am in a remote location.

6by9, 2017-09-28:
Have you disabled the Linux frame buffer? I don't see it listed in your output of ```vcgencmd dispmanx_list```, but was that from your copy/paste?
Removing that would reduce the overall screen complexity and therefore free up some HVS resource.

ephestione, 2017-09-28:
I didn't do anything to this raspbian installation in regard to the frame buffer (I wouldn't know how), should I disable that though? And how would I go about that?

In the meantime, I confirm that dividuum solution worked both on the screenshot side and the realtime hdmi display, no artefacts with two high resolution frames.

ephestione, 2017-09-28:
Also tested again with dispmanx_offline=1, with the monitor in front of me, rebooting and enabling the high resolution streams for two frames out of three. No artefacts on the screenshots nor the video, at least for the first 10 minutes or so, but after that, every other screenshot taken has artefacts.

closed 2019-11-05 by JamesH65