raspberrypi/firmware #882: vcdbg sym does not work as expected

paolosabatino, 2017-10-02:
vcdbg sym command does not do what is described in the binary embedded help. Issuing

`vcdbg help sym`

you get this description:

> Usage: vcdbg sym sym SYMBOL - Prints out the values of the indicated VC_DEBUG_VAR symbol

but then using vcdbg on any debug symbol variable like:

`vcdbg sym current_logging_level`

you get an output which is like this:

> Address: 0xdec02038 Size: 4

and the actual value(s) of the register is missing

This happens on raspbian updated to latest bits with update-rpi

Firmware:
> Aug 22 2017 14:18:53 
> Copyright (c) 2012 Broadcom
> version eb51b8b84ec11d98562dfbb7be53cdd216c0ea4c (clean) (release)

kernel:
> Linux raspberrypi 4.9.47+ #1033 Mon Sep 4 21:40:54 BST 2017 armv6l GNU/Linux

pelwell, 2017-10-02:
I think it does exactly what it says it does - you just need to understand the difference between a symbol and a variable. The value of a variable should be fairly clear, but a symbol (with a few exceptions) is effectively a pointer to a variable or function, so the value of the symbol is the address, not the content of the memory at that address.

paolosabatino, 2017-10-02:
Ok I think I got it, in fact to show the content of the address of the symbol directly `dump` option can be used:
`vcdbg dump current_logging_level`

I was actually misunderstanding the purpose of the command because the **syms** command has this description:
> vcdbg syms syms  - Prints out the values of the available VC_DEBUG_VAR symbols

and reports such table:
> Address          Size Value            Symbol
> 0x1f00b754         4 0x00000000 cma_stats_offline_count
> 0x1f00b758         4 0x00000000 cma_stats_online_count
> 0xdec02038         4 0x40010000 current_logging_level
> 0xdec0203c         4 0x00000000 dc4_stat_ptr
> 0xdec02040         4 0x00000000 display_status

where the Value field is the content of the address of the symbol.
The issue seems clarified enough to me, so I close.
Thanks

closed 2017-10-02 by paolosabatino