raspberrypi/firmware #883: Rfcomm bluetooth issues

subspacefield, 2017-10-03:
I have a Pi 3 running Stretch and a rfcomm server (C++) running on it. I am able to connect to it from my Windows and send messages to a camera that is connected to the USB port of the Pi. One of the "messages" is to get a frame from the camera. I can send camera frames from the Pi to the client using rfcomm for a few minutes before it seems that the Bluetooth on the PI locks up or crashes.

I say that it locks/crashes up because I have to reboot the Pi to connect to it again over Bluetooth. I can see the Pi over Bluetooth but you can not connect it once the failure occurs. I tried restarting the bluetooth service but that did not fix it. Then I tried "sudo hcoconfig hci0 reset" but get a timeout connecting.

I thought the problem might be the size of the data so I send in in 1024 byte chucks. The problem only occurs when I try to send a large amount of data from the Pi to the client.

If more details are needed or if this should be posted as a Linux issue let me know.

pelwell, 2017-10-12:
A new firmware is available via rpi-update that may have fixed this issue (there were two UART driver bugs) - it's been playing music to a Bluetooth headset in a loop for 16 hours with no issues.

subspacefield, 2017-10-16:
I tested retrieving images from camera attached to Pi  for several hours with no issues.

pelwell, 2017-10-16:
Please close if you are happy with the reliability now.

closed 2017-10-16 by subspacefield