raspberrypi/firmware #884: Feature Request: support kernel decompression in the bootloader firmware

aswild, 2017-10-07:
Unlike arm, the Linux kernel doesn't support the self-decompressing zImage format for arm64, and it's unclear whether there's intention to do so.  Instead, Linux arm64 expects the bootloader to decompress the kernel before jumping to it.

This means that 64-bit kernels are stuck being ~13MB uncompressed, which takes up SD card space and time to load the bigger image from flash. (Image.gz and Image.lzma are only 4.8 and 3.6 MB, respectively)  Using U-Boot is an option, but that adds several seconds of extra overhead to the boot time.

Would it be possible for the Pi firmware to support decompressing the kernel image before launching it? It could be controlled by a config.txt setting which specifies the compression mode.

pelwell, 2017-10-07:
From a performance perspective it doesn't make sense for the VPU to do the decompression. Fortunately it is now possible to create custom "stub" files - the initial instructions run by the ARM before control is handed to the kernel.

My preference would be to create a "super-stub" that decompresses the kernel before executing it. In practical terms this would be just another file on the SD card (armstub8.bin), and it would be up to the user to choose the appropriate stub for the compression mode.

pelwell, 2017-10-09:
The only other documentation that exists is this [extensive thread](https://github.com/raspberrypi/firmware/issues/579) in which the details were thrashed out.

Although I'm not aware of anyone using larger stubs yet, there is no hard restriction as far as the firmware is concerned provided the firmware-provided parameters are at the expected offsets.

aswild, 2017-10-10:
Thanks for the link, that was very extensive indeed.

I admit I'm new to low-level ARM boot code, but between that thread and the existing armstubs/u-boot implementations it sounds like enough info try taking a shot at a custom decompressor stub in my spare time.

If my understanding is correct, the flow of a decompressing stub could look like:

1. Do the same low-level CPU setup as armstub8.S. (primary CPU does everything below, others wait on their spin tables)
2. Set up the cache and stack so that C code can run
3. Read the kernel and dtb addresses provided by firmware in 0xfc and 0xf8
4. Pass those addresses to C code which
    1. Verifies the compression magic/header and decompresses the first block to obtain the Linux kernel's load address offset and size
    2.  Finds a suitable place to decompress the kernel to without overlapping the compressed data or dtb
    3. Decompress the kernel image (and memmove() it if a suitable offset couldn't be found)
    4. Returns the start address of the kernel
5. Assembly code turns off the cache and branches to the kernel start address

I see that @swarren's [start.S](https://github.com/swarren/rpi-3-aarch64-demo/blob/master/64/start.S) sets up the stack by just setting `sp` before jumping into C code. Would the stack pointer need to be saved and restored before booting Linux, or will Linux reinitialize it so it can be ignored?

I tweaked the demo files to print the original value of `sp` and it looks like it's initialized to a different but consistent value for each CPU between 0x37bed372 and 0xeef3efdf. Is that safe memory to use for the stack?

Thanks for all your help.

aswild, 2017-10-11:
**BLUF:** It looks like the fimware supports armstub8.bin being larger than 256 bytes on disk, but either truncates the stub to 256 bytes, or overwrites memory past 0x100 after loading the stub.

Last night I did some experiments trying to combine Stephen's C code and the armstub8.S code and get it all to run from the stub. I tried all sorts of things and different values for the kernel load address (including 0xffffffff in the stub to say not to load a kernel) but couldn't get anything to load.

Today I scaled back to a more basic task: prove that code past 0x100 can run in the stub, and it looks like this might not work on the latest 2cd872f firmware.

* I changed the initial values of x1-x3 and read them back from the "kernel" code to verify that my custom stub is being used.
* Then I moved the [`primary_cpu`](https://github.com/raspberrypi/tools/blob/master/armstubs/armstub8.S#L117) and `boot_kernel` blocks down to the "dtb_space" at 0x100. Those sections of code are only ever executed via labeled branches, so there should be no problem with that.
* I checked the disassembly to verify the stub magic and kernel/dtb load address fields are in the proper locations

Despite this, I still get no UART output from the kernel (the only way I can determine a sign of life on the board)

OK, so I can't seem to run code past 0x100 in the stub. What about data?

* I went back to the original (working) stub code
* added `.word 0xdeadbeef` to the `dtb_space` at 0x100
* changed the second-to-last line in `boot_kernel` to `ldr w3, dtb_space`

The kernel executes this time, but the value it gets from `x3` is 0x00000000_00000005, not my expected magic value.

aswild, 2017-10-11:
Update: I found you mention a "[gap](https://github.com/raspberrypi/firmware/issues/579#issuecomment-210990868)" in the other thread between 0x100 and 0x4000.

I found that if I move my extended code up to 0x4000, it can run! :D

What's that gap at 0x100 used for, is that were ATAGs are placed when a DTB isn't used?