raspberrypi/firmware #885: Pi Firmware could detect exFat better and warn end-user on screen

paul-hammant, 2017-10-08:
Like: direct them to a web page.  Or simply say that NOOBS should be put on a 32GB FAT partition and not a larger exFAT partition.

ghollingworth, 2017-10-08:
Except the Pi firmware is on the SD card and it can't read the SD card because it's exFAT formatted which means it can't load the code to warn you that you've formatted the SD card incorrectly...

closed 2017-10-08 by ghollingworth

paul-hammant, 2017-10-10:
So burn it into silicon then for Pi 4 and further revisions of 0 thru 3.

deltabeard, 2017-10-11:
If the partition is formatted incorrectly then that's on the user for not following instructions correctly.

lurch, 2017-10-12:
@paul-hammant I think you overestimate how much space there is inside the silicon, and underestimate how much code is needed to display a message on the screen.

@ghollingworth This also (sort of) links to https://github.com/raspberrypilearning/software-guide/issues/13