raspberrypi/firmware #886: SD Card incompatibility

bazzaxiv, 2017-10-09:
I decided to trial an industrial grade SD card and discovered that it will not boot on Pi 2 or Pi 3 devices. I have proven to my satisfaction that it is not the card or the image on the card see [https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=194848&p=1219773#p1219773]. The card can be found here [https://swissbit.com/data/view.php?q=S-45u/S-45u_fact_sheet.pdf]. I have ordered another card (they are not cheap) to confirm it is definitely an incompatibility issue and will post back here once I have it.

pelwell, 2017-10-09:
Have you seen issue #838 ?

bazzaxiv, 2017-10-09:
I missed that, thanks for the notification. I'll report this over there as it seems to the same problem but in my case it is 'hard' as in it does not 'reset' and never boots.

closed 2017-10-09 by bazzaxiv