raspberrypi/firmware #887: Bluetooth sound adpater stops working after few minutes

rnrmi, 2017-10-10:
I have some related issues pairing and using a bluetooth speaker (JBL GO). The device can be paired only after a reboot or before any management with bluetooth (GUI or CLI). Although the pairing "works" it is unable to remain functional after some seconds (or minutes) of audio stream (youtube or local sound), the connection is not lost but the communication between the rpi bt and the speaker seems to stop.

I'm using a clean Raspbian desktop stretch image recently update e totally functional, default configs, no overclock or altered configs.

pelwell, 2017-10-10:
I found and fixed one bug in the UART driver last week and enabled a workaround for another problem - together they improved Bluetooth reliability. I've just found another bug that can lead to data loss, the fix for which is not finalised, but with a workaround in place I've found UART data transfer to be very reliable and expect Bluetooth to be significantly better.

I'll comment again when a firmware containing all fixes is available.

asavah, 2017-10-10:
@pelwell I can confirm this too. Didn't reported it earlier because I'm using my own homegrown os and I currently don't have time to reproduce this with raspbian. 
For me it happens like this:
connect a2dp headset, start watching/listening to something using Kodi from popcornmix's newclock5 branch, after a random period of time from a few seconds to 30-40 mins the sound stops and bluetooth stack seems to die. After that even `hciconfig -a`  produces an error.
In dmesg ACL timeouts can be observed, I can't remember the exact messages right now. Only a reboot revives bluetooth. I've tried building bluez from git, tried both pulseaudio and bluez-alsa, tried different kernel branches, tried old and new bluetooth firmware blobs, tried all config options google could find - nothing seems to help.

If you have any ideas and/or patches we could try I could help testing as I can reproduce it easily.
On a side note: a cheapo usb bluetooth 4.0 dongle with CSR chipset works like a charm.

pelwell, 2017-10-11:
The second bug has now been patched, but it only affects devices using RTS/CTS flow control on ttyAMA0 - this includes Pi 3B rev 1.3 and Pi Zero W.

If you are running on an older Pi 3B then the current rpi-update firmware includes a fix for the only problem I am aware of. The flow control fix will be in the next firmware - can we roll one out tonight, @popcornmix ?

pelwell, 2017-10-11:
There is a fix to the kernel pl011 driver that went in this afternoon. It should affect a smaller number of people than the insertion of the memory barrier - just 3b1.3 and 0w - so we could wait until a few more patches have accumulated, but it does make a significant difference.

popcornmix, 2017-10-11:
Okay I'll try to remember when I get home.
@P33M was also keen on pushing a firmware update for pwm audio improvements.

pelwell, 2017-10-12:
A new firmware is available via rpi-update (thanks, @popcornmix) - it's been playing music to a Bluetooth headset in a loop for 16 hours with no issues.

rnrmi, 2017-10-12:
@pelwell & @popcornmix thanks people!

I can confirm that the issue is MUCH LESS problematic than it was. However, the issue seems to persist after 2 hours of music/video streaming. It works without any issue during the 1st hour, but when coming to the 2nd the BT device mute and I cannot disconnect or re-pair it to rpi. A reboot is needed to have a functional pair + connection + streaming.

I've tested mostly in browser (Chromium) but the issue seems to persist locally too.

fenyvesi, 2017-10-22:
Stretch Lite updated, upgraded. rpi-update done.

Bluetooth problems:
1. Playing is muted after 2-3 hours although the BT headset is connected yet.
2. The sound is terrible half of the time, the syslog says (several hundred times):
            bluealsa[496]: /usr/bin/bluealsa: BT socket write error: Resource temporarily unavailable

Any idea?

Thank you.

kle99, 2017-12-03:
The speaker is paired, trusted and connected via desktop controls. After some time the sound is aborted.

I'm using a clean raspbian stretch desktop image, default configs, ssh and vnc are enabled.
After some time seconds, minutes or hours (last time it was 15min), the speaker is silent but remains connected.

model: RPi 3B
audio: bluetooth speaker JBL GO (20cm distance to the Raspi)
os: 2017-11-29-raspbian-stretch.img
internet connection via internal WLAN (1m distance to the router)
RealVNC to Win7 notebook
ssh to Win7 notebook
test stream: http://st01.dlf.de/dlf/01/128/mp3/stream.mp3

journalctl:
Dec 03 20:14:01 raspberrypi systemd[1]: Starting Cleanup of Temporary Directories...
Dec 03 20:14:01 raspberrypi systemd[1]: Started Cleanup of Temporary Directories.
Dec 03 20:16:22 raspberrypi bluealsa[505]: /usr/bin/bluealsa: BT socket write error: Resource temporarily unavailable

bluetoothctl:
[JBL GO]# info <device mac>
Device <device mac>
	Name: JBL GO
	Alias: JBL GO
	Class: 0x240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
       ...

michpern, 2017-12-27:
There still seems to be an issue.  I am running the latest firmware/stretch version.   I am not trying to play sound continuously:  I am basically using the speaker as an alarm clock - I play a short sound and then wait a number of hours before trying to play a sound again.   After a reboot, playback is fine.  If I try the next day then there is no sound.  bluetoothctl/info on the device confirms that it is connected.

uname -a
Linux raspberrypi 4.9.70-v7+ #1068 SMP Mon Dec 18 22:12:55 GMT 2017 armv7l GNU/Linux

rnrmi, 2018-03-09:
Hi,

the issues persists. After recent updates the "connection lost" problem is happening very often.

Tyebeca, 2018-05-16:
I have the same issue. Is there any solution for this?

symetryczny, 2018-10-09:
Experiencing the same issue. Using LibreELEC, the sound drops after anywhere from 2 to 40 minutes. Bluetooth adapter disappears from the system and only a reboot fixes the problem.

j-whitehouse, 2019-01-13:
I'm not OP, but I can confirm I was having this issue with my RPi 3 B+ (fully updated) while streaming a netradio station with SMPlayer. I say was, because I have now been playing audio for a few hours straight cleanly by following this post https://github.com/raspberrypi/linux/issues/1402#issuecomment-405844346. In short, updating to `bluez-5.50` fixed the issue. Now to sort out multiplexing :)

4-3is4-me, 2019-01-14:
> Can any participents try the latest kernel/firmware and confirm if this issue is still present?

The issue is still present for me. Update ran 13 January and then the BT speaker sound dropped after about 7 minutes. Unable to reconnect until reboot.

rnrmi, 2019-01-14:
Hi,

I've reinstalled everything, updated raspbian and the issue persists. I can use a BT speaker for sometime, be it 2h ou 10 minutes, but suddenly (randomly) it disconnects and I have to reboot the system to reconnect -- until it loses connection again.

ram94638, 2019-01-17:
I have RPI 3B, I have paired bluetooth device on raspberry pi, Neither connect automatically to paired device nor provide serial port to device. I always connect manually(GUI). To connect by GUI I installed blueman package, I search everything on internet but no solution, Can help someone?

pelwell, 2019-01-17:
@ram94638 Please create a new issue.

jhdgghost, 2019-01-19:
Just tested in my rpi3b+ with librelelec 9 build 2019-0119. The a2dp audio disconnect after 5-15 minutes

vflam, 2019-06-04:
I have the same issue using Raspbian or Librelec, bluetooth stops working after 5-15 minutes and I have to reboot.

kle99, 2019-06-06:
> Can any participents try the latest kernel/firmware and confirm if this issue is still present?
test configuration: 
--- same as in my post from 3. Dec 2017 is: ---
model: RPi 3B
display: none (via vnc)
audio: bluetooth speaker JBL GO (20cm distance to the Raspi)
internet connection via internal WLAN (1m distance to the router)
RealVNC to Win7 notebook
ssh to Win7 notebook
test stream: http://st01.dlf.de/dlf/01/128/mp3/stream.mp3
--- different is: ---
os: Raspbian Stretch with Desktop (image 2019-04-08-raspbian-stretch.img)
uname -a
Linux raspberrypi5 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
---
test on 05.06.2019 from 19:10 until 22:10 bluetooth connection ok
test on 06.06.2019 from 09:55 until 13:20 bluetooth disconnected at 13:20
---
journalctl:
Jun 06 13:20:04 raspberrypi bluealsa[562]: /usr/bin/bluealsa: BT socket write error: Connection timed out
Jun 06 13:20:04 raspberrypi bluetoothd[556]: Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (1
Jun 06 13:20:04 raspberrypi bluealsa[562]: /usr/bin/bluealsa: Couldn't release transport: GDBus.Error:org.freedesktop.DBus.Error.UnknownObject: Meth

`pi@raspberrypi:~ $ date`
`Do 6. Jun 13:36:41 CEST 2019`

`pi@raspberrypi:~ $ sudo bluetoothctl -a`
`[NEW] Controller <controller-mac> raspberrypi [default]`
`[NEW] Device <device mac> JBL GO`
`Agent registered`
`[bluetooth]# info <device mac>`
`Device <device mac>`
`        Name: JBL GO`
`        Alias: JBL GO`
`        Class: 0x240404`
`        Icon: audio-card`
`        Paired: yes`
`        Trusted: yes`
`        Blocked: no`
`        Connected: no`
...

sbappudi, 2019-08-26:
Same issue on both a Pi 4 and a Pi Zero W, both on Raspbian full and on Diet pi. Very annoying error.

ernstwys, 2019-09-04:
Same here.. anybody noticed consistency with physical changing position of the devices.. I am noticing better signal (less disconnections) when having less objects between receiver and sender .. specific iron once; disturbing the connection... still looking for good solution though. 

Ideally would be a script for:

Reboot when audio disconnect + Auto connect bluetooth speaker on boot

pelwell, 2019-09-16:
A fix for a UART driver bug that could cause data-loss was fixed in July. For people seeing Bluetooth disconnection problems, please try the following:
```
$ sudo apt-get update
$ sudo apt-get install raspberrypi-kernel
$ sudo reboot
```

jacoghi, 2020-01-09:
Issue is still present for me as well, it's incredibly annoying cause there's no consistency at all.

gmazilla, 2020-01-19:
Have the same issues: after few minutes BT speaker stops speaking:(
if try to reconnect it says:
> Failed to connect to a device GDBus.Error.org.bluez.Error.NotReady Resource Not Ready. Try to connect again.

```
[57107.608142] input: E8:07:BF:FE:8C:6B as /devices/virtual/input/input5
[57213.028121] input: E8:07:BF:FE:8C:6B as /devices/virtual/input/input6
[57764.117837] input: E8:07:BF:FE:8C:6B as /devices/virtual/input/input7
[57857.634793] input: E8:07:BF:FE:8C:6B as /devices/virtual/input/input8
[58507.060607] Bluetooth: hci0: hardware error 0x00
[58511.500932] Bluetooth: hci0: command 0x1003 tx timeout
[58513.580988] Bluetooth: hci0: command 0x1001 tx timeout
[58515.660985] Bluetooth: hci0: command 0x1009 tx timeout
[58519.821012] Bluetooth: hci0: hardware error 0x00
[58521.901106] Bluetooth: hci0: command 0x1003 tx timeout
[58523.980982] Bluetooth: hci0: command 0x1001 tx timeout
[58526.061041] Bluetooth: hci0: command 0x1009 tx timeout
[58530.061200] Bluetooth: hci0: hardware error 0x00

```

Tested with different BT speakers.
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
```

Ruminateeee, 2020-01-22:
Issue seems to occur with me as well. I noticed keeping the Bluetooth menu open gives the connection a few more hours. Eventually it just gives out. Has anyone found a fix yet?

ernstwys, 2020-01-22:
I put the the RPI closer to the speakers. That helped a lot for me. So no object between those two.

refeved, 2020-02-17:
Exactly the same problem here. Bluetooth connection will lost after several minutes playback, and the program will lockup when try to stop the stream. 
The log of dmesg:
Bluetooth: hci1: hardware error 0x00
Bluetooth: hci1: command 0x1003 tx timeout
Bluetooth: hci1: command 0x1001 tx timeout
Bluetooth: hci1: command 0x1009 tx timeout

The log of bluealsa:
bluealsa: D: ../../src/ba-transport.c:1020: Releasing transport: A2DP Source (SBC)
bluealsa: E: Couldn't release transport: GDBus.Error:org.freedesktop.DBus.Error.UnknownObject: Method "Release" with signature "" on interface "org.bluez.MediaTransport1" doesn't exist
bluealsa: D: ../../src/ba-transport.c:1200: Exiting IO thread: A2DP Source (SBC)
bluealsa: D: ../../src/ba-transport.c:1020: Releasing transport: A2DP Source (SBC)
bluealsa: E: Couldn't release transport: GDBus.Error:org.freedesktop.DBus.Error.UnknownObject: Method "Release" with signature "" on interface "org.bluez.MediaTransport1" doesn't exist

I was using bluez-alsa on a Raspberry Pi 3B+ running Ubuntu-mate. The Bluez version is 5.48 from the depot originally. I have tried to upgrade Bluez to 5.52, but can not fix the problem.

When the issue happens, bluetooth will go off and stop responding completely. And I can not turn on bluetooth anymore, below is the output of bluetoothctl:
[bluetooth]# power on
Failed to set power on: org.bluez.Error.Failed
Restarting bluetooth service or bluealsa will also not help, the only way to recover is a reboot.

pelwell, 2020-02-17:
Unless Ubuntu Mate is taking patches from these raspberrypi/linux branches then it won't include my UART fix from a few weeks ago.

refeved, 2020-02-17:
> Unless Ubuntu Mate is taking patches from these raspberrypi/linux branches then it won't include my UART fix from a few weeks ago.

Thanks for your reply. It seems the fixed have already been submitted to raspberrypi/linux, right?

I have just reported an issue on https://bugs.launchpad.net/ubuntu-mate/+bug/1863600, hope Ubuntu mate can include the change in the future.

popcornmix, 2020-02-17:
This is the commit that avoids occasional lost uart characters and so may help bluetooth connectivity:
https://github.com/raspberrypi/linux/commit/65aa6ec0faaa012508489886ac357cbb86cdb9a4

refeved, 2020-02-19:
> 
> 
> This is the commit that avoids occasional lost uart characters and so may help bluetooth connectivity:
> [raspberrypi/linux@65aa6ec](https://github.com/raspberrypi/linux/commit/65aa6ec0faaa012508489886ac357cbb86cdb9a4)

I have tried to build and install the latest rpi-4.19.y branch of https://github.com/raspberrypi/linux on My 3B+ running Ubuntu-Mate 18.04. And confirm that it fixed the Bluetooth problem. I leave it play audio overnight and cannot see an disconnection. Thanks very much for the effort.
However, I'm not sure if it is safe to use raspberry/linux kernel directly on Ubuntu-Mate? I notice the the directory architecture and file name is different. I manage to get it work follow https://www.raspberrypi.org/documentation/linux/kernel/building.md and change some install steps. Well, the kernel works fine apparently.

pelwell, 2020-02-19:
This patch needs to be upstreamed - I think it's in a reasonable shape.

ernstwys, 2020-02-19:
Is this fix also for Raspbian ?

refeved, 2020-02-19:
> 
> 
> Is this fix also for Raspbian ?

As my understanding, raspberry/linux are meant to be run on Raspbian. I'm using Ubuntu-Mate, so I may have to wait the Ubuntu-Mate community to accept the patch. According to pelwell's reply, you can upgrade your kernel on Raspbian to get the fix directly:
$ sudo apt-get update
$ sudo apt-get install raspberrypi-kernel
$ sudo reboot

gmazilla, 2020-02-25:
> $ sudo apt-get update
> $ sudo apt-get install raspberrypi-kernel
> $ sudo reboot

RPI3B+, confirm problem fixed.

ChristofferGreen, 2020-05-24:
I am running Raspberry pi 4 with raspbian. Running:
`$ sudo apt-get update
$ sudo apt-get install raspberrypi-kernel`

gives: raspberrypi-kernel is already the newest version (1.20200212-1).

I still have the problem of bluetooth streaming of music stopping after a few hours

bjo81, 2021-11-06:
@ChristofferGreen Could you paste the output of `uname -a`?

fenyvesi, 2021-11-07:
This was a fast response. 18 months.