raspberrypi/firmware #890: md5sum of "archive" files has changed

bocadillodeatun, 2017-10-16:
This is a strange "issue". I'm posting it here in case anyone has an idea of what migth be going on.

The OpenWrt/LEDE project (older versions, at least) uses this repository to download snapshots of the rpi firmware.

The command used to download a specific snapshot is the next one:
```
wget https://github.com/Hexxeh/rpi-firmware/archive/<commit_hash>.tar.gz
```

Example:
```
wget https://github.com/Hexxeh/rpi-firmware/archive/38aa676b044f8de46aedb4bd972538a7ad6a3ce1.tar.gz
```

After downloading, the MD5SUM of the file is checked against a known value to make sure there were no errors while downloading the file.

Different versions of the OpenWrt/LEDE project download different commits and check against different values. These are some of them:
```
filename (https://github.com/Hexxeh/rpi-firmware/archive/)     expected md5sum                    
===============================================================================================
38aa676b044f8de46aedb4bd972538a7ad6a3ce1.tar.gz              f5683c1dcb255714942f7c9fd61b3a0a   
d4945b3b77d29cc5bb3777734422c048c1f1d003.tar.gz              18fe2d11acf9e277f7b3fb2cd18eabfb   
ab7a48a014a52af09df2f7a298bf36d545d3c1fe.tar.gz              842648dcb2f3e86f4ec5c85310f81d75   
```

Well... the last time I built an (old) OpenWrt image (two months ago?) these values matched the actual MD5SUM of the downloaded files. *However*, for some reason, this does not seem to be the case any more:

```
expected md5sum                    actual md5sum
===================================================================
f5683c1dcb255714942f7c9fd61b3a0a   edefa7a1684d5b0a2b11acd058adceff
18fe2d11acf9e277f7b3fb2cd18eabfb   a377312203096de291e2108e5f700304
842648dcb2f3e86f4ec5c85310f81d75   10d40dec1b9cfb35d627f224f4a9a730
```

Has "something" changed in this particular repository that migth justify this? (I'm asking because all other repositories OpenWrt depends on work... it's just this particular one were the issue is triggered).

I know it's not a "git history rebuild" or else the commit hashes would not match... it has to be something else. Have the repository "settings" (if that makes sense) been changed in the last two months or so?

Thanks.

PS: Feel free to close this issue if you can't think of a possible cause of this behaviour.

pelwell, 2017-10-16:
Duplicate of https://github.com/raspberrypi/firmware/issues/873 ?

bocadillodeatun, 2017-10-16:
Ups... you are rigth. Sorry. I should have tried to search a bit more for a similar issue before posting it.

closed 2017-10-16 by bocadillodeatun