raspberrypi/firmware #891: Boot from USB-SSD only works once

LFlare, 2017-10-17:
Hello,

Recently, I've made the switch from using SD cards to a SSD drive through USB for various reasons and the first thing I noticed that upon data migration from the SD card to the SSD drive, my **Raspberry Pi 3 Model B** server only manages to boot up once. Any subsequent reboots, shutdowns-and-boot, power-off-and-on fails to bring up the OS after the initial boot.

I tried to find out what was causing the issue and I think I narrowed it down to possibly the unmounting sequence during poweroff using `shutdown` or `reboot`. Basically, in order for me to reboot the server, I have to manually connect the SSD drive to my Macbook, mount the two partitions, unmount the two partitions, eject the drive, reconnect the drive to the Pi 3B and plug the power cable back in.

ghollingworth, 2017-10-18:
What about if you put in an SD card with _just_ bootcode.bin on it?  Also try adding a file TIMEOUT to the SD card as well...

Gordon

LFlare, 2017-10-19:
I tried that before, it still doesn't work on subsequent reboots. If it helps, I'm using the `Samsung T5 Portable SSD` without any encryption or whatnot. For now, my temporary workaround this issue is to simply separate the `boot` partition onto a `SD card` that points the `root` to my SSD drive.

ghollingworth, 2017-10-19:
It sounds like the drive is being left in some strange state...  Does
plugging it into a Raspberry Pi running Raspbian also 'fix' the problem or
is it only the Mac that can do it?

LFlare, 2017-10-19:
No clue, I've mitigated this by having the boot partition stay on a SD card.

smokingwheels, 2017-10-23:
I tried these steps and it worked. I was using a small HDD.
[https://www.maketecheasier.com/boot-up-raspberry-pi-3-external-hard-disk/](https://www.maketecheasier.com/boot-up-raspberry-pi-3-external-hard-disk/)
The MICRO SD card must remain in the Pi.

closed 2017-10-30 by LFlare

Draganis, 2018-03-19:
I experience the same problem. Mounting the SSD to a MAC, Linux or Windows system and unmounting it fixes the problem for just one boot. Any following subsequent boot shows up with the same error again. Putting an SD card into the device  will not solve the problem for me. This whole SSD thing was meant to be a replacement for the SD cards!

Draganis, 2018-03-19:
And it seems i have found something.
I was using an USB3 to SATA adapter with only one USB  plug to connect my SSD to the pi when i encountered the error.
Now i was trying to plug this adapter into a seperately powered USB hub that is connected to my pi.
Finally my pi reboots without having this issue. Seems like the SSD gets shut off cold without being properly unmounted during the shutdown sequence when it is not seperately powered. I also bought a new double plugged USB2 to SATA adapter to plug in one plug (data) to the pi and one plug (power) to a charger and the problem didn't show up either. So i guess this whole issue is a powering problem.

I am trying a cable like this one to bridge it: http://amzn.to/2IAO6rV