raspberrypi/firmware #892: cannot set OTP bit program_usb_boot_mode=1 with latest raspbian lite

alexheinz, 2017-10-23:
Hello,

with the latest raspbian lite image, it is not possible to set the bit to enable USB / network boot.

Tried it with the kernel that comes with raspbian and also after running `sudo BRANCH=next rpi-update` without success.

Had a backup of an old raspbian installation and with this it worked.

please also see [https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=195931](url)

ghollingworth, 2017-10-24:
You cannot set the OTP bit on anything but the BCM2837 device, you do not need this bit set to enable USB / network boot on a Pi0/1/2 if you are booting with an SD card containing just bootcode.bin

alexheinz, 2017-10-24:
Thank you for the feedback.

Has it always been like this? I am wondering because I do not remember exactly if network boot ever worked for me on a Pi1 without this bit being set. (I could set the bit on a Pi1 with an old raspbian.)

What about a Pi3 with an SD card containing just bootcode.bin (to enable booting from a TFTP server in a different subnet)? Do I need to set the OTP bit or not in this case?

I do not remember reading something about this in the documentation either, so unless I am mistaken, I would like to propose to update the documentation about network/MSD booting and clearly state that setting the OTP bit is not required on a Pi0/1/2 and maybe Pi3 if an SD containing just bootcode.bin is used.

ghollingworth, 2017-10-27:
Sure, submit a PR for the docs

Gordon

closed 2017-11-13 by ghollingworth