raspberrypi/firmware #895: SD with bootcode.bin fails with more than 1 USB storage device

HawaiianPi, 2017-10-28:
If I have a non-bootable USB flash drive connected at the same time as my boot USB-SSD, an SD card with only bootcode.bin fails to boot the system with 4 flashes of the green LED (which I believe means it can't load start.elf, so it's apparently checking the wrong drive).

Is that expected behavior? Can bootcode.bin only work with 1 USB storage device connected?

Procedure I used:
1. Format an SD card with the FAT32 file system
2. Copy the latest next branch bootcode.bin to the SD card
3. Insert SD card into my Pi3 and power on the system

Observed activity lights on both USB devices, but no boot, and green LED flashes 4 times over and over again. Adding a timeout file to SD card with bootcode.bin makes no difference.

Removing the extra USB flash drive results in SSD booting as expected.

The SSD can also boot with no SD card, even with the other USB drive plugged in, and it also works with /boot on the SD card. It's just the bootcode.bin only boot that fails with an extra USB storage device.

ghollingworth, 2017-10-31:
Yes bootcode.bin is relatively limited at the moment...  It just looks for the first MSD and tries that one, might be able to take a look at it over the next few months, but nothing quicker at the moment

HawaiianPi, 2017-11-02:
Alright, thanks. I suspected that might be the case. Not a big deal, I was
just wondering, and I didn't get an answer when I asked on the forums.

It would be nice if the capabilities of bootcode.bin could be improved, but
I'd rather have pure USB boot working stable. Right now I find that USB
boot from a power off state works great, but restart and shutdown are
unreliable.

Although I suppose that's a different issue altogether, so you can close
this report.

Thanks again.

closed 2017-11-02 by popcornmix