raspberrypi/firmware #896: [help] update vgfont/font.c to be compatible with utf-8

nagualcode, 2017-10-29:
I am trying to render fonts on the framebuffer, and I would ask for some help with the path to make vgfont/font.c compatible with utf-8.

paolosabatino, 2018-02-26:
it's not part of openvg beingcompatible with utf-8, openvg deals only with the rendering of the glyphs and provides a simple indexing of the path glyphs, but the decoding of the text string is not the purpose of the library

rodizio1, 2018-02-26:
I think this is not Raspberry firmware related.

You might want to check out Paeryin's fork of ajstark's OpenVG library available here.
The latest branch is the "windowsave" branch: https://github.com/paeryn/openvg/tree/windowsave

There is a test program included that displays every character found in a font, it also mentions UTF-8. You need to set locales in your system and inside your program accordingly.
https://github.com/paeryn/openvg/blob/windowsave/client/chars.c

nagualcode, 2018-03-09:
On https://github.com/raspberrypi/firmware/blob/master/opt/vc/src/hello_pi/libs/vgfont/font.c
There is:
#48   * This API doesn't support unicode at all at present, nor UTF-8.

And wy my tests hello_font example wont render latin acents.

paolosabatino, 2018-03-09:
@nagualcode no need to importunate popcornmix for this, that example is not part of OpenVG: it is an example that uses OpenVG.
As an exercise you may want to implement your own utf-8 decoder and apply it to that example

closed 2018-03-12 by nagualcode