raspberrypi/firmware #899: USB Mass Storage mount Raspberry in read-only mode

erotundo, 2017-11-07:
Hi!

In this case i follow the steps to enable mass_storage gadget into my raspberry pi zero w but when this connected to my computer mount the drive letter in read-only mode and i don't find way to disable this mode because i need read/write from OS

Regards,

Eduardo

ghollingworth, 2017-11-07:
What steps did you follow?

erotundo, 2017-11-08:
Hi!

I follow the basic step first

"/boot/config.txt" -> add dwc2
"/boot/cmdline.txt" -> add modules-load=dwc2,g_mass_storage after rootwait
"create mini filesystem" -> sudo dd if=/dev/zero of=/home/pi/usb-drive.img bs=1M count=50
"set permissions" -> sudo chown 755 /home/pi/usb-drive.img
reboot
connect to laptop with running Windows, after Windows can't install the component execute
"sudo modprobe g_mass_storage file=/home/pi/usb-drive.img stall=0"

With this step in Windows device manager appear "USB Mass Storage" but i can't Access to the "drive", the drive not appear in Linux

So i use another method to generate the mass storage
"/boot/cmdline.txt" -> add modules-load=dwc2,g_multi after rootwait
"/etc/modprobe.d/" -> create g_multi.conf and add "options g_multi file=/home/pi/usb-drive.img stall=0 removable=y ro=0 host_addr=11:22:33:44:55:66 dev_addr=aa:bb:cc:dd:ee:ff

With this last step, the drive appears into Linux and Windows read only mode and i need to modify the files into the minifile system drive

Regards,

Eduardo

ghollingworth, 2017-11-08:
This github issues list is for issues with firmware, the problem you are having is with an unsupported driver (dwc2) and you are following instructions written by someone else (Adafruit or gbaman), go to the forums for support or contact the writers of the instructions you have directly

closed 2017-11-08 by ghollingworth