raspberrypi/firmware #900: [Windows 10 IoT Core] Wake Up Raspberry PI by GPIO means

alexiordan, 2017-11-14:
I'm looking for a solution to wake up (after a software Shutdown) my Raspberry PI 3 Model B by GPIO means (shortcutting one GPIO pin to ground).
Can this be done inside bootcode.bin?

popcornmix, 2017-11-14:
The certain way is to short the RUN header which will reset the board and cause it to reboot.
Alternatively pulling GPIO3 low should also cause bootcode to resume booting if it is in halt state (i.e. you issued ``sudo halt`` and got the ten flashes on LED).

alexiordan, 2017-11-14:
1. I don't want the RUN method since I want to put this command on a button and switching it can also cause a hard-reset, which I don't want.
2. I am using Windows 10 IoT Core. I'm shutting it down by BeginShutdown(ShutdownKind.Shutdown). Being that said, I tried shorting GPIO3 to GND after that shutdown, and nothing happened.

popcornmix, 2017-11-14:
Did you get the ten flashes on ACT LED when you shutdown?
You'll only be able to wake it with GPIO if you did.

alexiordan, 2017-11-14:
I only get 7 ACUT blinks...

alexiordan, 2017-11-16:
On the other hand, is bootcode.bin platform specific? I mean is the one provided in Window 10 IoT image different from the one in Raspbian/Linux ?

pelwell, 2017-11-16:
There is only one sequence of firmware releases (containing one bootcode.bin and the four start.elf  variants), so Windows 10 IoT has to use them.

popcornmix, 2017-11-16:
On raspbian ``sudo halt`` results in 63 being written to the reset controller register and a watchdog reset being triggered. When bootcode.bin launches it checks for 63 in reset controller and enters a low powered halt state (before LAN, PLLs, clocks and power domains are enabled on the chip).
It will flash the ACT LED ten times at this point, it will go to sleep and only a GPIO event will allow it to continue (or a chip reset through RUN pin).

If you are not seeing ten flashes on the ACT LED then we aren't entering this low powered halt state.
Perhaps you should compare behaviour (and if it wakes up correctly) with raspbian.

If Windows 10 IoT does not enter the low powered halt state then you will need to ask the question there - modifying the firmware won't help.

alexiordan, 2017-11-16:
Thank you very much for this explanation.
I'll post this info on the Windows IoT MSDN forum.

mjmeans, 2017-11-16:
I too am only getting 7 ACT blinks, then a pause, then it repeats. If it is only blinking 7 times instead of 10, what state does this represent?

JamesH65, 2019-01-10:
This page explains all

https://raspberrypi.stackexchange.com/questions/41450/raspberry-pi-not-working/41455#41455

closed 2019-01-10 by JamesH65