raspberrypi/firmware #903: Remove stale dispmanx layers

carlonluca, 2017-11-24:
I have a system on which killing an application rendering OpenGL on a dispmanx layer sometimes result in what seems a "stale" dispmanx layer. The current frame being rendered when the kill arrives remains freezed on the screen and won't go away even after starting the process again. I used the command vcgencmd dispmanx_list and I see two layers with the same layer ID, one is the old one which seems to remain. Is there any way to clean the screen? Can I somehow remove that "zombie" dispmanx layer after my process is dead? Maybe with some other command or with dispmanx APIs? I understand that the application should cleanup before dying, but is there another way to guarantee proper cleanup even when the app cannot cleanup by itself? At the moment I cannot reproduce this simply with other applications. Killing does not typically seem to cause this.

ghollingworth, 2017-11-27:
Do you have an application that will reproduce the problem?

popcornmix, 2017-11-27:
If the dispmanx layer is not removed with the app then something bad has probably happened.
I suspect just removing the dispmanx layers (if it were possible) wouldn't really be a solution (there may well be OpenGL or other resources that haven't been freed up).
As @ghollingworth says, an application + script that starts/kills in a loop until the problem occurs would give the best chance of finding the problem.

JamesH65, 2019-01-10:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

carlonluca, 2019-01-11:
Yes, I think you can close for the moment. I tried to reproduce with a script a long time ago but I failed.

closed 2019-01-11 by JamesH65