raspberrypi/firmware #906: Can't boot mainline 4.14.0-rc3 using bcm2835-rpi-zero.dtb

notro, 2017-11-28:
config.txt
```
device_tree=bcm2835-rpi-zero.dtb
```

Result:
```
Uncompressing Linux... done, booting the kernel.

Error: unrecognized/unsupported machine ID (r1 = 0x00000c42).

Available machine support:

ID (hex)        NAME
ffffffff        Generic DT based system
ffffffff        BCM2835

Please check your kernel config and/or bootloader.
```

This however boots fine:
```
device_tree=bcm2835-rpi-b-plus.dtb
```

I used fdtdump and compared the two dtbs, it looked fine to me: https://gist.github.com/notro/c137159039ae26f690862821b1dc8c1a
Tried this on both a Zero and a B+.

I haven't tagged the kernel.

Kernel and dtb: http://tronnes.org/downloads/bcm2835-rpi-zero.tar.gz

pelwell, 2017-11-28:
What ought to be the same kernel and .dtb boots OK for me, but using yours I get the same error message.

Will investigate tomorrow.

pelwell, 2017-11-29:
The Pi Zero DTB contains no `/aliases` node, and the firmware code to add an alias doesn't handle this case (it never occurs in the downstream DTBs). In addition, this failure was being treated as if it were fatal.

There are two new patches in the firmware pipeline that create the `/aliases` node as necessary and suppress the non-fatal errors. I'll comment when the fixed firmware is available, but until then the obvious workaround should suffice.

notro, 2017-11-29:
Thanks Phil, adding the node solved it.
I see that the aliases node was added to bcm283x.dtsi in 4.14-rc6:
https://github.com/torvalds/linux/commit/f08f58a2bf68900a84e782b8c7ad701c0654173c

The drm subsystem tree has been stuck on 4.14-rc3 for some time, but I see that it moved to -rc7 a week ago so I'm in the clear when I fetch that. Thanks for helping me out.

pelwell, 2017-12-02:
The fix for this bug is in the latest firmware release.

closed 2017-12-02 by notro