raspberrypi/firmware #908: Change pixel order in firmware KMS mode

stschake, 2017-12-01:
Right now, firmware KMS is the only way to get the upstream VC4 graphics drivers and the touchscreen working simultaneously. The DSI panel works in 4.14 but then you don't get the touchscreen input anymore.

Unfortunately, the pixel order in the firmware KMS mode is hardcoded to RGB and can't be changed. That's fine for X but other systems prefer it flipped. I tried using the "set pixel order" property but the firmware isn't accepting that. 

Is it possible to introduce a way for signalling BGR to the firmware? Maybe the 30th bit in bpp similar to the way T-format tiled scanout works now. That would be perfectly backwards compatible and we could then advertise BGRA and RGBA to DRM.

popcornmix, 2017-12-01:
Can you give examples of code that wants RGB order flipped?
Are you planning on modifying the VC4 kernel drivers to support this?
Have you discussed this with @anholt? If I implement this I'd like it to be in a way he is also happy with.

stschake, 2017-12-01:
Android always defaults to BGR (it says _RGBA but that is flipped to the DRM definition):

https://android.googlesource.com/platform/frameworks/native/+/master/opengl/libs/EGL/eglApi.cpp#659

I'm happy to provide a patch for the firmware KMS that does this on the kernel side. It only involve advertising the BGR formats to DRM:

https://github.com/anholt/linux/blob/rpi-4.9.y/drivers/gpu/drm/vc4/vc4_firmware_kms.c#L319

And then telling the firmware in some form or shape, similar to tiled scanout:

https://github.com/anholt/linux/blob/rpi-4.9.y/drivers/gpu/drm/vc4/vc4_firmware_kms.c#L141

I haven't gotten hold of @anholt, I think hes out of office this week.

stschake, 2017-12-04:
I chatted with @anholt and while he is not opposed, he would of course prefer we use the full/non-firmware KMS. We'd prefer that too, but with current firmware, it seeems to incapacitate the 5406 touchscreen. I've read reports of people using the upstream DSI *and* the firmware touchscreen driver simultaneously - have there been changes to the firmware that disabled that?

closed 2019-11-05 by JamesH65