raspberrypi/firmware #909: vc4-fkms-v3d and vc4-kms-v3d not work on 4.14.y

amikvic, 2017-12-03:
Hi,

If i activate one of these overlays in 4.14.y branch, either the Pi doesn't boot (vc4-fkms-v3d) or the screen stays black (vc4-kms-v3d). I will try to post dmesg if necessary.

pelwell, 2017-12-03:
Loading the overlay is not enough - there is some extra configuration required. Use `sudo raspi-config` to enable the "Experimental OpenGL driver", and it will perform the necessary steps.

amikvic, 2017-12-03:
I'm not using Raspbian, i prefer Gentoo for several reasons.
Dmesg for vc4-kms-3d are these :
[    8.875141] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[    8.881902] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    8.888552] [drm] Driver supports precise vblank timestamp query.
[    8.889443] [drm] Cannot find any crtc or sizes

I don't know how config must be, but here my config.txt :
dtoverlay=vc4-kms-v3d
dtoverlay=pi3-disable-bt
#hdmi_safe=1
disable_overscan=0
config_hdmi_boost=6
hdmi_force_hotplug=1
gpu_mem=128
avoid_warnings=2
#hdmi_drive=2
dtparam=audio=on

fluffysheap, 2017-12-19:
It doesn't matter whether you use Raspbian or Gentoo or something else for this, since the problem occurs in the kernel, and it also occurs with the [official 4.14 beta kernel on Raspbian](https://jp.raspberrypi.org/forums/viewtopic.php?f=29&t=197689) in any case.  With KMS and the 4.14 kernel from Pi foundation it really is almost as if the DSI interface never existed.

In my attempt to troubleshoot the problem, I have observed that the usual line:
vc4-drm soc:gpu: bound 3f700000.dsi (ops vc4_dsi_ops [vc4])
is not present in dmesg.  The "Cannot find any crtc or sizes" is a red herring, it means no display was detected, but says nothing about why.  Similarly, the usual DSI files in /sys are all missing or empty.

I've also noticed that the usual DSI-oriented modules such as panel_raspberrypi_touchscreen and rpi_backlight are not loaded during this problem, but loading them manually does nothing.

The problem continues to occur with the latest (today's) 4.14.7-v7+ beta kernel, but I don't know what version of the KMS driver goes into that.

@amikvic You should probably have something like cma=128M@512 (first number should match your gpu_mem setting) and also disable_overscan=1 in your config.txt, but it won't make your display work in this case, sorry.

amikvic, 2017-12-27:
Hi,
Thanks for the answer.
For the simple kms driver... it was my fault, i'm fed up to do some stupid mistakes. Apparently, my hdmi was not plugged well. But it's a bit strange, before all was previously running well. I forced the cable a little in the plug and now i can run with kms driver.
For the fkms, the problem stays same, i don't have a serial cable to see why it doesn't boot.

fluffysheap, 2017-12-31:
I've poked around fkms a little.  In Raspbian, it gets past kernel boot and into systemd before it hangs. 
 Because of the "special" nature of systemd, in absence of a specific error message, there's not really any way to know what happened.  In Gentoo (no systemd) it actually gets to the filesystem check and hangs.  So I don't know much more there either.

This is not what I expected, but that's a common theme this month with these video drivers.  I can perhaps try single-user mode for more precise debugging, but now that I have true KMS working, I (probably, unless I find worse problems in KMS) have no further use for FKMS, so it would just be for curiosity.

JamesH65, 2019-01-10:
Please update to latest driver.

Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2019-01-10 by JamesH65