raspberrypi/firmware #912: Cannot Create Window Surface

thedigitalempress, 2017-12-06:
I am trying to complete the 2nd program in Chapter 6 of "Learning Python with Raspberry Pi" and after running the code on the pi I get this error in the stack trace. 

[image]

I've been researching for some time now on how to resolve the issue but haven't found much. I did analyze the rip_egl.py code and saw that I should come here if there are any issues. 

[image]

It will run the first half of the code and will not display the pygame window for the music player. This is my code. 

https://github.com/thedigitalempress/screenDance/blob/master/screenDance.py

Thanks for your help!

JamesH65, 2019-01-10:
Closing as this issue is not kernel related. Please use the forums for this sort of question.

closed 2019-01-10 by JamesH65

ghollingworth, 2019-01-10:
Just checking with the author of the book as to where you should post this question.

Thanks

Gordon