raspberrypi/firmware #913: Black screen with SDL1.2 binaries

mw-mbelge, 2017-12-13:
I am seeing the same issue, exactly as described in this thread, on 4.9.59-v7+. Basically, I get a black screen when running SDL1.2 binaries on a headless Pi 3. Has the same issue come back again?

pelwell, 2017-12-13:
1. Have you tried on earlier kernels? `sudo rpi-update <commit-id>` allows you to choose a specific version, where the commit-id can be found on the [Hexxeh firmware repo](https://github.com/Hexxeh/rpi-firmware/commits/master) (use the links on the right of each entry).

2. It's probably a silly question, but what does it mean to get a black screen on a headless Pi 3? Where is the screen?

mw-mbelge, 2017-12-13:
What I meant by headless is no desktop is installed (Lite version). Console is output to HDMI. SDL binaries used to work in this configuration through framebuffer drivers. 

I'll try earlier kernels see at what point SDL broke.

arielscarpinelli, 2017-12-19:
I'm experiencing the same issue with 4.9.41-v7+, so it is gotta be older than that

DFXThomas, 2018-03-19:
I can confirm this is still a problem with the latest Raspbian and a Raspberry Pi 3, Model B+.

JamesH65, 2019-01-09:
This appears to be a problem in the SDL 1.2 library, not the Pi firmware. However, the linked issues seems to say it is fixed in our repo prior to this report, so not entirely sure whats going on.  I'll have a look tomorrow to see if the patch has somehow fallen out. Ping @XECDesign just in case.

JamesH65, 2019-01-09:
(Of course, if anyone has tried a recent distro, and its now works, please add comment here)

XECDesign, 2019-01-10:
Thanks, I'll take a look.

XECDesign, 2019-01-10:
I've added patches from RetroPie's version of SDL and it works a lot better in general. Haven't checked different kernel versions.

@spl237 are the other changes staged in the repo good to go live? If so, I'll push those along with the sdl fixes.

XECDesign, 2019-01-10:
Updated sdl 1.2 should be in the repo now. Has it fixed the issue?

JamesH65, 2019-01-10:
Thanks for doing that @XECDesign, I cannot try it but hopefully someone on participents list can.