raspberrypi/firmware #914: Forcing USB device mode boot even when there is a full firmware on the SD card.

ali1234, 2017-12-13:
This is part question, part feature request.

Suppose I have a RPi Zero with a blank SD card, and I load the MSD firmware over USB with rpiboot or usbbootgui. Now I write my firmware on to the USB storage device and reboot. It should now boot my firmware from the SD card.

But supposing I write a bad firmware and the device goes in to a boot loop. Is it possible to force a boot from USB without removing the SD card? So that I can once again boot the MSD firmware from rpiboot and write a different firmware on to the SD card?

The N900 had a boot option called "boot_wait" for this purpose. It would specify number of seconds to wait for a USB connection before continuing normal boot. Would it be possible to have something like this in config.txt? (If it does not already exist.)

ghollingworth, 2017-12-15:
Unfortunately not, once the bootrom finds a bootcode.bin on the SD card it will boot from it.  This is why we specifically have to interrupt the CMD line from the eMMC to the processor when you want to boot from USB on the compute module (otherwise it'd be impossible to get out of this situation)

closed 2017-12-15 by ghollingworth

ali1234, 2017-12-15:
Wow. I'm glad I'm not trying to build a real product based on this.

JamesH65, 2017-12-15:
Because sarcasm works so well when wanting help.

ali1234, 2017-12-15:
Help has already been refused.

pelwell, 2017-12-15:
First of all, this paragraph makes no sense:

> Suppose I have a RPi Zero with a blank SD card, and I load the MSD firmware over USB with rpiboot or usbbootgui. Now I write my firmware on to the USB storage device and reboot. It should now boot my firmware from the SD card.

Secondly, "refused" implies a lack of will, when in fact it is a lack of possibility - the bootrom is immutable, and it does what it does.

Thirdly, if you were serious about building a field-updateable product then you would partition the SD card (assuming you aren't using an EMMC which is more suited to the job - Pi Zero was never intended to form part of products, that's what CM is for) and create a nominally read-only partition containing U-boot or some other loader - NOOBS even -  and put the OS/application in a separate partition.