raspberrypi/firmware #919: kernel: usb 1-1.3: reset high-speed USB device number 4 using dwc_otg

greencopper, 2018-01-04:
I started using a new RPI 3  today onto which I have attached a seagate USB hard drive.

Normally I have never seen any issues on the other PI's I got, but this one - from time to time - throws this in the log:

```
kernel: usb 1-1.3: reset high-speed USB device number 4 using dwc_otg
```
I have attached this drive: https://www.seagate.com/staticfiles/support/disc/manuals/notebook/momentus/5400.2/SATA/100367122c.pdf

To this box: http://www.raidsonic.de/products/external_cases/external_cases_25/index_en.php?we_objectID=3437

Are the power requirements for the drive too high?

I have run a Seagate Backup Plus Portable Drive (1TB) on a RPI 3 without any issues on the same PSU - a 5V 2.5A Raspberry PSU.

Also if it is a power problem, is it possible to solve it using a Y USB cable?

Kind regards

greencopper, 2018-01-18:
I added a small USB hub with an external PSU and the issue has stopped.

Before I added the hub one of those reset messages caused the system to go into "read only" mode.

These power issues on USB with Raspberry needs to be fixed. I also have a first gen. Cubox which is capable of running with two external USB drives attached without any problems on a 2A PSU, and it is able to boot right of the top USB drive. The USB is even gigabit.

jakemagee, 2018-01-19:
I think you discovered / solved the issue @greencopper (too much power draw).  As for fixing this on the Raspberry Pi, I'm not sure there is anything to fix.  That is a USB3 enclosure... and while it obviously will run as a USB2 device when connected to a USB2 host, it probably still draws USB3 spec'd power.  It looks like USB3 devices can draw up to 900ma and USB2 is only spec'd to deliver 500ma.

greencopper, 2018-01-22:
I just had two more issues today - with the harddrive attached to the externally power USB hub.

```
Jan 22 00:54:10 rp2 kernel: usb 1-1.3.1: reset high-speed USB device number 5 using dwc_otg
Jan 22 00:54:11 rp2 kernel: usb 1-1.3.1: reset high-speed USB device number 5 using dwc_otg
```

I'm about to give up on using R-PI3 for anything serious.

jakemagee, 2018-01-23:
You mentioned running the Seagate Backup Plus Portable Drive without any problems.  Have you tried extended testing with an x86 based machine and the new sata / enclosure combo (with the USB hub)?  Is it possible that this combo is having non-raspberry pi related issues?  I've seen other posts with these errors that stated their USB hub also did not supply enough power.

greencopper, 2018-01-24:
I purposefully bought an enclosure that many people recommended for Raspberry PI, but it might be that even the enclosure is having problems with this drive.

But I think I'm just gonna remove the Raspberry and use a Cubox instead, I have never had any issues or the need for an enclosure on the Cubox. I have run two external hardrives on the same Cubox on a 2A PSU without problems.

jakemagee, 2018-01-24:
Interesting... have any of the people that are using this combination responded as to wither they have ever seen the issue?  Of course, I assume your using the latest Raspbian release (I realize now I never asked that before)...

And I can't fault anyone for sticking with what works for them.  However, if your not interested in the issue any longer, could you close the ticket?

jakemagee, 2018-01-24:
In addition to making sure you are running the latest Raspbian release, have you tried setting "max_usb_current=1" in config.txt ?

Also, insufficient power supplies / cables are pretty common.  Have you tried using a higher output power supply or thicker/shorter cable?  It's common enough that an official 5.1V 2.5A power supply is being sold.

greencopper, 2018-01-24:
I am unable to contact "the other people", I found the information on different websites.

I am not running Raspbian, but Arch Linux, and I have the "max_usb_current=1" setting in config.txt.

The PSU is the standard 5.1V 2.5A with the Raspberry PI logo on it.

What "worries" me is that so many people have these issues and they always get "handled" as a PSU/power problem. Since I have worked with other PI like boards such as Cubox, Odroid, and others, and have never experienced any kinds of power issues on these board, with even several HD attached to USB, I cannot help to wonder why only the Raspberry boards seem to struggle so much with this issue? The other boards I have use even lower powered PSU's.

jakemagee, 2018-01-25:
Can you possibly re-try with the latest version of Raspbian (since its the "official" distro and could have kernel patches to address the issue)?

I think alot of similar issues get handled as PSU problems because it turns out to be the solution... at least early on with the RPi since alot of people would just assume any USB power adapter / cable would work fine (and I think why an official power supply was created).

I also saw a reference to a similar error message and the drive spinning down from inactivity.

ghollingworth, 2018-01-25:
The reason there are issues is that the 'USB' HDDs you are using do not meet the USB 2.0 specification, if they did then this would not be a problem.  From your linked specification it says the spinup current is 1.1 A.

The USB specification limits the supply current to 500mA for a high powered device.  If you use externally powered HDDs then it would not be a problem.  The Raspberry Pi hardware has a current limit device (set to 1.2A) to avoid the USB devices from browning out the rest of the board (and therefore instead of the error you are seeing the Pi would just reset instead).

closed 2018-01-25 by ghollingworth

greencopper, 2018-01-25:
Please do not close this issue as I am using an external powered HDD.

jakemagee, 2018-01-26:
I don't think there is anything that can be done (from a software side) for the issue.  That hub is a USB 2.0 hub (much like the one built into the RPi).  It is probably having an issue supplying enough power as well.  You might have better luck with a USB 3.0 hub that can supply 3.0 power requirements / specifications.  While the hub uses a 2A power supply, and might be capable of supplying close to 2A, that's most likely not to any individual port but spread across all 4 ports.