raspberrypi/firmware #921: Pi 3 Model B USB boot off MSD (with no SD card) not working with MyDigitalSSD 120GB

rkomorn, 2018-01-04:
I can't get my Pi 3 Model B to boot off of a USB drive without the SD card in it, or even with just bootcode.bin on a file.

Booting with an SD card works fine. Booting without the SD card just gives me a black screen and proceeds no further.

The drive is a MyDigitalSSD (the 120GB model on http://mydigitalssd.com/mobile-ssd.php#otgv2-usb-3.1 ).

I'm filling this issue because it's mentioned in the instructions on https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md . I'm not at all sure if this is because the drive is one of the unsupported devices, or because of an issue on my particular Pi 3.

I confirmed that the OTP's properly programmed:

```
root@raspi:~# vcgencmd otp_dump | grep 17
17:3020000a
```

The one thing I didn't exactly follow from the docs is that I'd already been booting a Pi 2 model B off USB on a spinning drive, so I replicated the partition table and dd'ed over files. I didn't just use etcher to copy the image onto the drive. The boot flag was my addition after the initial failure to boot.

```
root@raspi:~# fdisk -l /dev/sda
Disk /dev/sda: 119.2 GiB, 128035676160 bytes, 250069680 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x824d251f

Device     Boot Start      End  Sectors  Size Id Type
/dev/sda1  *     8192    93236    85045 41.5M 83 Linux
/dev/sda2       94208 62333951 62239744 29.7G 83 Linux
```

As mentioned, booting with only bootcode.bin on a FAT32-formatted card (see image below) doesn't work: I don't even get output to the HDMI port. The green LED alternates 3 flashes with a steady light.

[image]
 
bootcode.bin was downloaded around 10am PST on 2018-01-04 .

I'd love some help troubleshooting this, and some help making this issue report better.

pelwell, 2018-01-04:
Try adding a `timeout` file, as in #922.

rkomorn, 2018-01-04:
No difference in behavior. Still blinking three times, intermittently. No output to HDMI.

burtyb, 2018-01-05:
Looks like the filesystem on /dev/sda1 isn't FAT32 (or you have the wrong type set in the partition table).

rkomorn, 2018-01-05:
Oh, you're totally right. That just never mattered before since I'd resorted to mounting /boot from the SD card instead and the /dev/sda partition. /etc/fstab had it as FAT, so it was still mounting properly.

Thank you for catching that.

closed 2018-01-05 by rkomorn