raspberrypi/firmware #922: Can't boot with Pi 2 Model and SD card with bootcode.bin on it.

wasi54, 2018-01-04:
Flash Drive OCZ Vertex 4 SATA III Solid State Drive 64GB.
It is connected by a dual USB connector adaptor (one for data and one for USB power supply).
If the Power for  the Flash raises about 50-100 ms before the Power of the Raspi, then booting is fine and everthing is working as expected.
If the the power raises at the same time, it won't boot and the gren LED lamp on the Pi starts blinking after some seconds.
Is there a way to increase the delay the Pi start?

Thanks in advance

pelwell, 2018-01-04:
Create a file called `timeout` on the SD card and see if that helps. It may have to non-zero length.

wasi54, 2018-01-04:
Great, thats the solution!

closed 2018-01-04 by wasi54