raspberrypi/firmware #923: Post install / first boot script

jakemagee, 2018-01-05:
There was mention in the forums (https://www.raspberrypi.org/forums/viewtopic.php?t=190921) about a first boot script functionality (similar to /boot/ssh and /boot/wpa_supplicant.conf) coming eventually.  Is this still planned?  If so, could we keep a ticket open for tracking it?

I also assume there aren't any other cool /boot file features like these... are there?

jakemagee, 2018-01-11:
Another issue reference for a /boot/ "boot" script, https://github.com/raspberrypi/piserver/issues/3, although this request seems to be for a script that executes on every boot rather than only for the first boot.

popcornmix, 2018-01-12:
Sorry this isn't the right place for this (this repo concerns the GPU firmware).
https://github.com/RPi-Distro/repo may be a better place for this.

closed 2018-01-12 by popcornmix

jakemagee, 2018-01-12:
Your absolutely right... just because a file / functionality is in the /boot/ directory doesn't mean that a question belongs in this repo.  Sorry about that, I didn't think long enough before posting.

And thanks for pointing me in the right direction.  https://github.com/RPi-Distro/pi-gen/issues/47 existed and has already been closed as a low priority and needing better direction for the implementation.

Also, just as a note, the implementation for /boot/ssh and /boot/wpa_supplicant.conf is in https://github.com/RPi-Distro/ .  Along with code for /boot/os_config.json (Apply Noobs config).