raspberrypi/firmware #924: Crackling sound while playing AC3 via analog out

Quantomax, 2018-01-06:
Hi, 
since commit d7a6e668ca0fb79518fd1063d27e741fd300c60d (Oct 12, 2017) I get audio distortion, if I play video tracks with only AC3 audio via vdr + rpihddevice.

Sounds a bit like "hard clipping" or added fireplace crackling. 

Any hints?

JamesH65, 2019-01-09:
@Quantomax It's been a while, sorry, does running the latest firmware makes things any better?

Quantomax, 2019-01-09:
With 4.14.92 the issue is still present...
Conditions: Playback of 1920 x 1080 @50i Channels with AC3 only via PAL out + Analog audio.

I have to copy
- start_cd.elf, start_db.elf, start.elf, start_x.elf
and from 4.14.7x (don't remember) on additionally:
- bootcode.bin, fixup_cd.dat, fixup.dat, fixup_db.dat, fixup_x.dat
form the last working firmware, to make it work smooth.

BR,
QuantoMax

JamesH65, 2019-01-10:
@popcornmix Not sure who else to ask, but does this ring any bells?

popcornmix, 2019-01-16:
Does the problem occur with ``audio_pwm_mode=0`` in config.txt.

Quantomax, 2019-01-17:
With 4.14.93 and audio_pwm_mode = 0 AC3 works as normal.

Previous config.txt was (still not working):
audio_pwm_mode=2

P33M, 2019-01-17:
Does the AC3 decode happen on the VPU or is it performed by Linux in software?

Quantomax, 2019-01-17:
Hmm, good question...

I use rpihddevice, which uses ffmpeg libs to decode audio and omx-player for playback.
If I read
https://www.vdr-portal.de/forum/index.php?thread/126514-beta-rpi-ausgabeplugin/&postID=1260622#post1260622
correct, the AC3 to PCM down-mix is done by ffmpeg in software an than pushed to omxplayer for hardware accelerated output.

I think, if you take a look at the code at
https://projects.vdr-developer.org/git/vdr-plugin-rpihddevice.git/tree/audio.c
you'll see, which parts are really used.

P33M, 2019-01-17:
Can you try a 1080i video stream with PCM audio stream with and compare? I.e. is the crackling dependent on just the AC3 audio.

If no (you get crackles with both), then do you have any 720p or 1080p content that you can mux with the AC3 stream and see if that changes things? The audio doesn't need to match the video, so any random progressive content will do.

Quantomax, 2019-01-18:
Yesterday ServusTV transmitted 1080i with AC3 (5.1) and PCM, recorded it.

1920x1080i -> PAL-B/G // PCM -> Analog out: Sound a bit louder, but with cracks
1920x1080i -> PAL-B/G // AC3(5.1) -> Analog out, as above, but lower volume (center mix).
1280x720p -> PAL-B/G // AC3(5.1) -> Analog out -> Sound smooth.
720x576i -> PAL-B/G // AC3(5.1) -> Analog out -> Sound smooth

If it helps, I can make the recording available for download.

P33M, 2019-01-18:
So it looks like 1080i content is causing the audio dropouts.

Please do upload the video files.

popcornmix, 2019-01-18:
Is the decoder deinterlacing? Does disabling (or switching away from advanced deinterlace) avoid the issue?