raspberrypi/firmware #925: Odd screen artifacts

dividuum, 2018-01-10:
Hi.

I've a weird problem that maybe one of you can help me with: I've got a Pi connected to a [VIZIO M-Series 80"](https://www.vizio.com/m80c3.html) display through HDMI. During the night I'm using the following sequence of commands to turn off that display:

```
tvservice -o
echo standby 0 | cec-client -s
```

In the morning the display is powered back on using

```
echo on 0 | cec-client -s
tvservice -p
```

This works as intended but for some reason the video output sometimes looks like this:

[image]

Rebooting the Pi fixes the problem. I'm not sure what do to: Capturing the snapshot of what's on the screen using [my screenshot tool](https://github.com/info-beamer/tools/tree/master/screenshot), which uses the dispmanx API for that, doesn't show the problem: The output generated by the Pi seems to be fine. The exact same setup works well on an NEC and Samsung display. Any idea what might cause this?

Firmware/Kernel (I didn't try a newer version, but browsing trough the firmware repo commit log doesn't seem to have anything remotely related to that):

```
Sep 12 2017 15:21:04 
Copyright (c) 2012 Broadcom
version 3fbb171d1950eb3789eb8e37ba11e01079a37239 (clean) (release)

4.9.50-v7+ #1035 SMP Wed Sep 13 23:16:24 BST 2017 armv7l GNU/Linux
```

rodizio1, 2018-02-16:
Sorry if this doesn't help your case, but I have once tried to implement an "automatic HDMI detection" feature like in this post using tvservice commands: https://www.raspberrypi.org/forums/viewtopic.php?t=12332#p952362

It did not work reliably at all, got all kinds of strange issues, crashes, monitor display all garbled-up, etc. so I gave up on it.

popcornmix, 2018-02-16:
Very likely a combination of poor quality cable and the VIZIO display being a less sensitive than the other displays. If you have another HDMI cable try swapping.
Otherwise you can increase the drive to HDMI.
Try adding ``config_hdmi_boost=7`` to config.txt (you can increase it up to 11 if that doesn't help).
See https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

dividuum, 2018-02-19:
I'll try these suggestions and reopen this issue if the problem happens again. Thanks.

closed 2018-02-19 by dividuum