raspberrypi/firmware #926: RPi-3 Register Conflict (local_intc vs. syscon)

kevans91, 2018-01-13:
Hi,

There seems to be a register conflict in bcm2710-rpi-3-b.dtb at least, shared between /soc/local_intc and /soc /syscon@40000000. Is this intentionally shared between the two? I've not found any documentation to suggestion that this should be the case, and /soc/syscon doesn't seem to exist at all in Linux DTS.

pelwell, 2018-01-16:
Yes, there is a conflict that should be fixed. Is it currently causing a problem or is this just an observation?

The syscon node is there to support the FIQ handling on BCM2836/BCM2837. It was added in the 4.4 tree, which predates the arrival of the local_intc node upstream. @notro kicked our downstream DTS files into shape back then, and he may be able to give some informed comment on its continued necessity, but for some uninformed comment...

It appears that the syscon subsystem is being used as a mechanism for a driver to locate and make use of a regmap used by another device in the DT. Doing so requires that the node with the regmap is declared to be compatible with syscon, but I don't think it needs anything more than that. One option may be to add the syscon compatible string to local_intc in the downstream DTS, and to change the compatible string passed as a parameter to `syscon_regmap_lookup_by_compatible` - `"brcm,bcm2836-armctrl-ic"` ought to work. Another might be to not use syscon at all, since it seems we are using it as a convenience function, and to find some other way of sharing the regmap, but its almost certainly more bespoke code.

kevans91, 2018-01-16:
This was mostly from an observation standpoint- it did cause a problem, but not one that can really be blamed on it. =)

I added a generic syscon driver to FreeBSD and botched the attach order, so it attached before local_intc and broke some things because syscon takes exclusive access of registers and other use is expected to pass through this specific driver (generally).

This left me wondering why the syscon node was there given that it wasn't in Linux and nothing else appeared to be referencing it as I would have expected from other boards that I work with. Your explanation seems to make sense to me, though, so thanks!

I'll await comment by @notro or others, but it does seem to me that it would make sense to add the syscon compatible to the local_intc to eliminate the syscon node, if that would work from a Linux perspective.

notro, 2018-01-16:
Yes, syscon is about sharing registers between drivers.
Since irq-bcm2835.c and irq-bcm2836.c are both built-in and both selected together by CONFIG_ARCH_BCM2835, it's possible to add a function to irq-bcm2836.c instead and call that from irq-bcm2835.c avoiding the regmap.
If that is done, I suggest to revert my 2 patches first so we have only one downstream patch to handle FIQ. It makes for easier reading. It was difficult to squash together the two patches in my mind to understand what was going on :-)

pelwell, 2018-01-16:
Thanks, notro - that makes a lot of sense, and is probably what we will do.

kevans91, 2018-01-16:
Ah, so that's the situation where I'm now used to seeing something like:

1.) syscon is labelled node
2.) No regmap in intc, but a "syscon = <&syscon>;"
3.) Driver uses that to get regmap instead it being specified via 'regs' in DTS

notro, 2018-01-16:
> it's possible to add a function to irq-bcm2836.c instead and call that from irq-bcm2835.c avoiding the regmap.

Or just use a global variable for the register.

pelwell, 2018-01-23:
I've pushed a patch to rpi-4.9.y that removes the syscon node and regmap use. I tried reverting your patches, notro, but it got ugly and most of what you did is retained so the new patch is just an additional tweak.

notro, 2018-01-23:
Nice to get the syscon use out of the way. I believe it was added because we had 2 ARCH'es at the time and irq-bcm2836 wasn't built on 2708.

> I tried reverting your patches, notro, but it got ugly and most of what you did is retained so the new patch is just an additional tweak.

Maybe it can be done on a later version.

I like how we now have downstream patches per driver/subsystem, which makes it easy to see the difference to mainline, at least as long as rebasing is done.
I scan through the downstream patches when a new version is released to get a rough view of how things are.

Maybe this one: [Main bcm2708/bcm2709 linux port](https://github.com/raspberrypi/linux/commit/f0da9b731576627e8d7cebf4baa327f008ca0b51), could be split up and labeled accordingly per change as well for 4.15 or later :-)

At least the syscon select can be removed with your change.