raspberrypi/firmware #928: VCSM dmabuf sharing & tiling modes

stschake, 2018-01-22:
This one is a bit "out there". So @6by9 recently made importing dmabufs into VCSM possible, and that's working great to have the camera zero-copy into a mesa-vc4 texture. However, the data is dumped linearly, and as I understand it, the GPU only ever supported RGBA linear textures and even that support is broken. So right now, after all the MMAL stuff is done, we still have to do a full copy from the linear texture into a tiling format so that the hardware can actually sample from it, with the associated memory & CPU usage.

Since the GPU-side must already be doing some conversion internally to support RGB565 and the other formats it can output, is it possible to have it write the data in T-format tiling?

6by9, 2018-01-22:
Not easily, and certainly not in hardware.

The only 2 hardware blocks that handle T-format are the V3D and HVS, and having just scanned the HVS docs the use of T-format there is nominally deprecated.
The video codec writes out a column based YUV420 format. The JPEG block always does planar YUV 420/422. The ISP writes out numerous raster formats, or the codec column format. None of the other hardware does T-format. Any conversion would therefore have to be done on the VPU, which is little gain over doing it on the QPUs.

What's the failure with RGBA linear textures? I remember conversations with regard that for the firmware GLES driver, and I got the impression it was feasible, if not done.

stschake, 2018-01-22:
The errata list for V3D has HW-2898 ("Address generation error for NPOT raster textures"). I guess having to always force power of 2 widths makes it annoying enough not to bother, certainly the mesa v3d driver always allocates a tiling shadow for linear/raster textures.

I might have a closer look into what the proprietary driver is doing when it imports linear YUV.

6by9, 2018-01-22:
For 1080P video then the power of 2 isn't such a big deal (1920 -> 2048). For 720P it is rather wasteful.
I'm just getting up to speed on graphics - is it Mesa that sets up the tiled shadow buffer, or the kernel driver? Pointers most welcome :-)

The proprietary driver does the conversion to RGB T-format on the VPU, at least when using the MMAL path described at https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiTex.c#L57

stschake, 2018-01-22:
Mesa sets that up:

https://github.com/bminor/mesa-mesa/blob/master/src/gallium/drivers/vc4/vc4_state.c#L575

6by9, 2018-01-22:
Thanks for the pointer.
It sounds like the MMAL parameters could be set up fairly easily to make it fit within the restrictions of HW-2898, but that's for another day.

Very few of the multimedia hardware blocks are set up for RGBA at all. The one where it would be a fairly minimal overhead to create RGBA_TF is vc.ril.isp - the hardware can produce RGBA planar, and then the VPU can reorder fairly easily to TF. It'll also connect happily to video_decode etc to do the YUV to RGB format conversion in hardware. There's still an SDRAM bandwidth hit, but less processor time.

JamesH65, 2019-01-09:
@6by9 Intending to do anything on this, or shall I "This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested."

6by9, 2019-01-09:
Close in 30 days.
It's so far down the list that it'd need significant use case to warrant it. I believe Eric has written conversion shaders for most of these formats anyway.

stschake, 2019-01-09:
I'll go ahead and close it, with the YUV stuff Eric has done for mesa, this is pretty much obsolete anyway.

closed 2019-01-09 by stschake