raspberrypi/firmware #929: CONFIG_USB_TMC=m

jakemagee, 2018-01-22:
Would it be acceptable to add USBTMC (CONFIG_USB_TMC=m) to the list of pre-compiled modules?  I doubt it gets used much, but it would keep some of us from having to build and distribute it manually.

pelwell, 2018-01-22:
What does it do? You aren't exactly selling it. ;-)

Ferroin, 2018-01-22:
USB TMC is the USB Test and Measurement Class.  Example devices include USB oscilloscopes, USB logic analyzers, and USB connected multimeters.  Essentially, it's the USB equivalent of the old SCSI and RS232 interfaces a lot of test-bench equipment used to push data to a computer.

pelwell, 2018-01-22:
It's a 25KB module - not the smallest but not exactly bloated. I'm OK with including it - @popcornmix ?

popcornmix, 2019-01-10:
I'm okay with this being added if @jakemagee (or others) would still find it useful.

dedekind, 2019-01-11:
+1
It is useful, please, enable it.

dedekind, 2019-01-11:
I have users trying to control Yokogawa power meters from Raspberry Pi, and today 'yokotool' (https://github.com/intel/yoko-tool) requires the usbtmc kernel driver for that, so having the driver available as module would help the users.

janzhen96, 2019-01-14:
+1 
Can you please enable the usbtmc kernel driver.

pelwell, 2019-01-14:
Added - see https://github.com/raspberrypi/linux/commit/b3a4c69c56f4db99b091305a33aa7c9b090aaad1.

popcornmix, 2019-01-14:
Available in latest rpi-update kernel.

popcornmix, 2019-04-01:
@jakemagee please test and close if happy.

JamesH65, 2019-11-05:
Closing this issue as questions answered/issue resolved.

closed 2019-11-05 by JamesH65