raspberrypi/firmware #931: RPi3 with Suptronic X850- Boot problem from SSD

pepeEL, 2018-01-24:
Hi
I buy to my raspberry pi3 module name Suptronic X850 to connect SSD mSATA to my Raspberry. I use disc mSATA ADATA SP310 32GB.
I connect all and install on my SSD image disc RASPBIAN STRETCH LITE. After connect to my RPI3 all works ok and boot ok. System started... but when i done command:
sudo apt-get update
then 
sudo apt-get upgrade 
then
sudo reboot
After commend reboot system not boot. When unplug power and connect again system boot ok. Again done command 
sudo reboot 
and again system not boot. Again unplug power again connect and works ok. Problem is still after reboot rpi3. Where is the problem.
After failure boot green LED on Suptronic X850 light very weak light. When works ok light very good light.
Please help me...
I see that suptronic x850 has chip Jmicron and my disc SSD ADATA SP310 also have chip JMicron. It is a problem ?

pepeEL, 2018-01-25:
Anybody can help me please...

pelwell, 2018-01-25:
Am I right in thinking that both the USB<->SATA adapter and the SSD are powered from a single USB port? That places quite a load on the Pi power supply - probably well outside the USB spec of 500mA for a high-powered device.

Can you try with a powered hub between the Pi and the adapter to see if that makes a difference?

pepeEL, 2018-01-25:
But i dont have option to connect to other power hub...
Look
http://www.suptronics.com/miniPCkits/x850.html

pelwell, 2018-01-25:
I think that's what the blue cable is for - plug one end into the adapter and one into the hub, then use a standard USB A (rectangular) to USB B (square) cable to plug the Pi into the hub's upstream port. Don't use the small circuit board with two USB plugs on.

pepeEL, 2018-01-25:
ok i try...

pepeEL, 2018-01-25:
Ok i cut power from USB and get from GPIO 5V to power USB Suptronic X850. After this all works ok but on firmware 4.9.59-v7+ 
After update firmware to 4.9.78-v7+ again after reboot not run no boot.
Please help me or correct problem in next new firmware.

pepeEL, 2018-01-26:
Can you help me anybody how i can come back to oled FW ?

pelwell, 2018-01-26:
The hash value has to be taken from the [Hexxeh/rpi-firmware](https://github.com/Hexxeh/rpi-firmware/commits/master) repo, but then just run `sudo rpi-update <hash>` as before.

pelwell, 2018-01-26:
P.S. Updating a comment does not send any notifications. If you are adding new information or asking new questions it is better to add a separate comment.

pepeEL, 2018-01-26:
Ok thanks for help... I try it today

pepeEL, 2018-01-26:
But if i want install new FW from other branch name NEXT... how command i must use ? ANd what command to comeback to old master branch ?

pelwell, 2018-01-26:
`sudo BRANCH=next rpi-update` is another way of saying `sudo rpi-update <hash-for-the-head-of-the-next-branch>` (and `sudo rpi-update` is an alias for `sudo BRANCH=master rpi-update`), so if you know the hash then just run `sudo rpi-update <hash>`. The hash identifies the firmware, and the Pi has no interest in which branch the current firmware comes from, so switching branches is just a matter of installing a new firmware from the branch you want, i.e. `sudo rpi-update` "switches back to" the master branch.

pepeEL, 2018-01-26:
ok very very thanks good help :)

c64emulator, 2018-03-18:
I have several mSATA-USB-Adapters for the Raspberry Pi: the PiDRIVE from Kickstarter (Prolific PL2571B), the Lycom-PI102 (Renesas uPD720231A) and the X850 (JMicron JMS578). Only the X850 has the restart problems  (sudo reboot). I  own several mSATA-SSD too (from Samsung, Micron, Dell, Sandisk). All this mSATA-SSD work with other adapters without problems.
I got  the X850-v1.3, with a 5V-connector. Using the connector does not solve the issue.
I testet with Pi3 and Pi2-v1.2 with current raspbian kernel (4.0.x) and kernel from branch=next (4.14.x).

pepeEL, 2018-03-18:
Hi
Today i resolved this problem. I flash my X850 with new FW and now all works ok.
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=196334&p=1287803#p1287803

jakemagee, 2018-03-20:
@c64emulator , have you tried updating the X850 firmware to see if that fixes your issue as well?

c64emulator, 2018-03-20:
I have testet the JMS578_Hardkernel_v173.01.00.01.bin ([https://wiki.odroid.com/odroid-xu4/software/jms578_fw_update](https://wiki.odroid.com/odroid-xu4/software/jms578_fw_update)) and the JMS578_v255.01.00.01_beta.bin ([https://forum.odroid.com/viewtopic.php?f=97&t=29069#p211459](https://forum.odroid.com/viewtopic.php?f=97&t=29069#p211459)). Neither of them fixes the issue for me.
Even a sdcard with the most current bootcode.bin doesn't do it for me. I have to replug the power connection to the raspberry after "sudo reboot" to bring the RPi to life again.
The bridge controller (JMS578) of this converter board seems to be crappy. There is already a patch for drivers/usb/storage/unusual_devs.h in the linux mainline kernel ("usb-storage quirk").

pepeEL, 2018-03-20:
Today i flashed by FW JMS578_v255.01.00.01_beta.bin  and after that my X850 stop working and is not found under lsusb.
Also connect it to PC under Winsows and also not found.
I try reflashed it to JMS578_Hardkernel_v173.01.00.01.bin but i get error:

`pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# cd JMS578FwUpdater
root@raspberrypi:/home/pi/JMS578FwUpdater# ./JMS578FwUpdate -d /dev/sda -v
ERR : open device fail
Get Bridge version FAIL!

root@raspberrypi:/home/pi/JMS578FwUpdater#`

`root@raspberrypi:/home/pi/JMS578FwUpdater# ./JMS578FwUpdate -d /dev/sda -f ./JMS578_Hardkernel_v173.01.00.01.bin -b ./backup.bin
Update Firmware file name: ./JMS578_Hardkernel_v173.01.00.01.bin
Backup Firmware file name: ./backup.bin
ERR : open device fail
Get Bridge version FAIL!

ERR : open device fail
Read Back to Backup Error!!
root@raspberrypi:/home/pi/JMS578FwUpdater#`

I try flash it by programmer USB CH341 like this:
[https://www.dhresource.com/0x0s/f2-albu-g5-M01-E2-01-rBVaI1jYgAOAHtb5AAFFauQAnRo674.jpg/ch341-ch341a-usb-programmer-sop16-dip16-sop8.jpg](url)

but in software version 1.29 i can not found CHIP Pm25LQ010. Only found Pm25LV010

please help me...
In Pm25LQ010 is stored Firmware ?

pepeEL, 2018-03-21:
@c64emulator  after flash FW make do this:
[image]

For me help this. Without this also not working after reboot.

c64emulator, 2018-03-21:
I own the hardware revision v1.3. This version has already an extra 5V connector:
[image]