raspberrypi/firmware #932: Come back to older FW

pepeEL, 2018-01-25:
Hi
I update my FW in RPi3 to latest by command: Sudo rpi-update
I use SSD mSATA on my RPi3 by adapter Suptronic X850 , and after update to latest FW my SSD after reboot not boot. Before works ok on FW 4.9.59.
I want back to 4.9.59 but i dont know how because long hash not working with command sudo rpi-update <long hash>

I also ask how i can go to to branch 4.14.x ? and then go back to 4.9.x

pepeEL, 2018-01-26:
Can you help me anybody ?

popcornmix, 2018-01-26:
See: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=197689 for info on 4.14 kernel.

For downgrading, see:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Reboot after this completes.