raspberrypi/firmware #933: License information missing for boot/ files

agraf, 2018-01-26:
The boot/ directory contains a few files that have their license described in /README.md, but most of the files are not covered by any license and there is no explicit "fallback" license for the overall project.

Please either provide a project fallback license or explicitly mention the licensing terms for all boot/ files (like overlays, dtbs, etc).

pelwell, 2018-02-08:
`/modules`, `/debug`, `/extra` (except `dt-blob.dts`), the dtbs, overlays and associated `README` are built from source files in the kernel tree, and are covered by the same licence as the kernel images. `/extra/dt-blob.dts` and `documentation/ilcomponents` are, like the start.elfs, covered by `LICENCE.broadcom`.

I think this should cover all files now, but let me know if there are still omissions.

`README.md` will be updated accordingly.

pelwell, 2018-02-08:
See PR https://github.com/raspberrypi/firmware/pull/938.

agraf, 2018-02-08:
I think with the updated text, all "fixup*.dat" files are still not covered, right?

"loader.bin" does not appear in the /boot directory for me. I assume that file used to exist and got removed?

As for all GPL licensed files (kernel, dtbs, overlays) you need to refer to the exact code that produced these binaries. I assume that means you'll have to note the sha1 hash of your own kernel tree every time you update them. Also, are you 100% sure the overlays are generated from your kernel tree? Can you point me to the exact location?

Thanks!

afaerber, 2018-02-08:
> I think with the updated text, all "fixup*.dat" files are still not covered, right?

...as already pointed out in my merged #498 commit. ;)

pelwell, 2018-02-08:
You're right about the wildcard, fixup*.dat and loader.bin - I'll fixup dat. @afaerber I don't know how your change got lost - don't take it personally. 8-)

I don't feel we have to refer to the exact code that produced the GPLd binaries - it is all in the Linux source tree (apart from the C compiler), and that should be enough. And yes I am 100% sure the overlays are in our kernel tree - see https://github.com/raspberrypi/linux/tree/rpi-4.9.y/arch/arm/boot/dts/overlays/ (the dtb sources are one level up).