raspberrypi/firmware #934: PXE Problems with RPi3

DenSA-Inc, 2018-01-28:
Trying to configure a Raspberry Pi 3 to boot from LAN has put me through several hardships. Before I describe the problem(s) I explain my setup first.

I have two pi's, one Pi 3 which I want to boot from LAN, and one Pi 2B which acts as router for the network. Both pi's are connected to a switch by ethernet-cables. The router is configured to automatically set its IP-address for the network (10.0.0.1, using the 10.0.0.0/24-network-segment) and is running dnsmasq exclusively on that network. The Pi 3 is configured to boot via PXE, running `vcgencmd otp_dump | grep 17:` prints `17: 3020000a`. I made sure of that several times.

Now to my problem. First of all, it takes the pi ~1 hour to send a DHCPDISCOVER-packet. I don't know why this can happen, but it makes testing much more tedious. Second, no matter what I try the Pi 3 does not accept my DHCPOFFER-packet as an invitation to boot. I've tried multiple configurations, I even compiled dnsmasq myself to use the newest version which accepts the `dhcp-reply-delay`-option.

This is my dnsmasq-configuration:

    port=0
    dhcp-range=10.0.0.10,10.0.0.111,12h
    log-dhcp
    enable-tftp
    tftp-root=/cloud/tftpboot/
    pxe-service=0,"Raspberry Pi Boot"
    dhcp-boot=bootcode.bin
    dhcp-host=b8:27:eb:bc:a9:07,10.0.0.10
    dhcp-reply-delay=1

This is a dissection of the two packets sent between the two pi's:
    No.     Time        Source                Destination           Protocol Length Info
         68 4254.801076 0.0.0.0               255.255.255.255       DHCP     362    DHCP Discover - Transaction ID 0x26f30339
    
    Frame 68: 362 bytes on wire (2896 bits), 362 bytes captured (2896 bits)
    Ethernet II, Src: Raspberr_bc:a9:07 (b8:27:eb:bc:a9:07), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
    Internet Protocol Version 4, Src: 0.0.0.0 (0.0.0.0), Dst: 255.255.255.255 (255.255.255.255)
    User Datagram Protocol, Src Port: 68 (68), Dst Port: 67 (67)
    Bootstrap Protocol (Discover)
        Message type: Boot Request (1)
        Hardware type: Ethernet (0x01)
        Hardware address length: 6
        Hops: 0
        Transaction ID: 0x26f30339
        Seconds elapsed: 0
        Bootp flags: 0x0000 (Unicast)
        Client IP address: 0.0.0.0 (0.0.0.0)
        Your (client) IP address: 0.0.0.0 (0.0.0.0)
        Next server IP address: 0.0.0.0 (0.0.0.0)
        Relay agent IP address: 0.0.0.0 (0.0.0.0)
        Client MAC address: Raspberr_bc:a9:07 (b8:27:eb:bc:a9:07)
        Client hardware address padding: 00000000000000000000
        Server host name not given
        Boot file name not given
        Magic cookie: DHCP
        Option: (53) DHCP Message Type (Discover)
            Length: 1
            DHCP: Discover (1)
        Option: (55) Parameter Request List
            Length: 12
            Parameter Request List Item: (43) Vendor-Specific Information
            Parameter Request List Item: (60) Vendor class identifier
            Parameter Request List Item: (67) Bootfile name
            Parameter Request List Item: (128) DOCSIS full security server IP [TODO]
            Parameter Request List Item: (129) PXE - undefined (vendor specific)
            Parameter Request List Item: (130) PXE - undefined (vendor specific)
            Parameter Request List Item: (131) PXE - undefined (vendor specific)
            Parameter Request List Item: (132) PXE - undefined (vendor specific)
            Parameter Request List Item: (133) PXE - undefined (vendor specific)
            Parameter Request List Item: (134) PXE - undefined (vendor specific)
            Parameter Request List Item: (135) PXE - undefined (vendor specific)
            Parameter Request List Item: (66) TFTP Server Name
        Option: (93) Client System Architecture
            Length: 2
            Client System Architecture: IA x86 PC (0)
        Option: (94) Client Network Device Interface
            Length: 3
            Major Version: 2
            Minor Version: 1
        Option: (97) UUID/GUID-based Client Identifier
            Length: 17
            Client Identifier (UUID): 44444444-4444-4444-4444-444444444444
        Option: (60) Vendor class identifier
            Length: 32
            Vendor class identifier: PXEClient:Arch:00000:UNDI:002001
        Option: (255) End
            Option End: 255
    
    No.     Time        Source                Destination           Protocol Length Info
         69 4255.812073 10.0.0.1              10.0.0.10             DHCP     401    DHCP Offer    - Transaction ID 0x26f30339
    
    Frame 69: 401 bytes on wire (3208 bits), 401 bytes captured (3208 bits)
    Ethernet II, Src: Raspberr_a6:13:1b (b8:27:eb:a6:13:1b), Dst: Raspberr_bc:a9:07 (b8:27:eb:bc:a9:07)
    Internet Protocol Version 4, Src: 10.0.0.1 (10.0.0.1), Dst: 10.0.0.10 (10.0.0.10)
    User Datagram Protocol, Src Port: 67 (67), Dst Port: 68 (68)
    Bootstrap Protocol (Offer)
        Message type: Boot Reply (2)
        Hardware type: Ethernet (0x01)
        Hardware address length: 6
        Hops: 0
        Transaction ID: 0x26f30339
        Seconds elapsed: 0
        Bootp flags: 0x0000 (Unicast)
        Client IP address: 0.0.0.0 (0.0.0.0)
        Your (client) IP address: 10.0.0.10 (10.0.0.10)
        Next server IP address: 10.0.0.1 (10.0.0.1)
        Relay agent IP address: 0.0.0.0 (0.0.0.0)
        Client MAC address: Raspberr_bc:a9:07 (b8:27:eb:bc:a9:07)
        Client hardware address padding: 00000000000000000000
        Server host name not given
        Boot file name not given
        Magic cookie: DHCP
        Option: (53) DHCP Message Type (Offer)
            Length: 1
            DHCP: Offer (2)
        Option: (54) DHCP Server Identifier
            Length: 4
            DHCP Server Identifier: 10.0.0.1 (10.0.0.1)
        Option: (51) IP Address Lease Time
            Length: 4
            IP Address Lease Time: (43200s) 12 hours
        Option: (67) Bootfile name
            Length: 13
            Bootfile name: bootcode.bin
        Option: (58) Renewal Time Value
            Length: 4
            Renewal Time Value: (21600s) 6 hours
        Option: (59) Rebinding Time Value
            Length: 4
            Rebinding Time Value: (37800s) 10 hours, 30 minutes
        Option: (1) Subnet Mask
            Length: 4
            Subnet Mask: 255.255.255.0 (255.255.255.0)
        Option: (28) Broadcast Address
            Length: 4
            Broadcast Address: 10.0.0.255 (10.0.0.255)
        Option: (60) Vendor class identifier
            Length: 9
            Vendor class identifier: PXEClient
        Option: (97) UUID/GUID-based Client Identifier
            Length: 17
            Client Identifier (UUID): 44444444-4444-4444-4444-444444444444
        Option: (43) Vendor-Specific Information (PXEClient)
            Length: 32
            Option 43 Suboption: (6) PXE discovery control
                Length: 1
                discovery control: 0x03
            Option 43 Suboption: (10) PXE menu prompt
                Length: 4
                menu prompt: 00505845
            Option 43 Suboption: (9) PXE boot menu
                Length: 20
                boot menu: 00001152617370626572727920506920426f6f74
            PXE Client End: 255
        Option: (255) End
            Option End: 255

Logs from dnsmasq are not available because I can't get the dnsmasq-service to work with the self-compiled dnsmasq.

Two problems are to solve: finding a reason for the pi's late boot and getting it to accept the DHCPOFFER. The last one might have to do with the line "Boot file name not given" wireshark gives me, even though I specified it in the options.

popcornmix, 2018-01-29:
Ping @ghollingworth

ghollingworth, 2018-01-29:
I would think the problem is to do with your network switch ignoring the first few packets transmitted from the Raspberry Pi...  To check you could copy bootcode.bin to an sdcard (on it's own with nothing else) and boot from there, if it then works then it's one of the bugs we know about in the bootcode.

  DHCP is transmitted 5 times then fails, some network switches take longer than this to enable the port and therefore don't work.  (disabling spanning tree protocol usually helps here)
  If we receive a DHCP _and_ the TFTP response within 2 seconds the device will get locked into a massive loop.  To get around this add dhcp-reply-delay=2 (I don't know whether this means you cannot have a single dnsmasq serving both the IP address and the TFTP server address.)

Otherwise I'd suggest running PiServer on your device, it should be able to reliably boot the Raspberry Pi clients.

DenSA-Inc, 2018-01-29:
@ghollingworth Ok, I'll try. Does the sd-card need a special fs (fat for example, as it is the boot-section-fs)?

ghollingworth, 2018-01-30:
FAT32 should be fine

DenSA-Inc, 2018-02-02:
@ghollingworth Starting a Pi with a sd-card which only contains bootcode.bin (I simply erased everything else from a sd-card where I dd'ed a raspbian-image onto) gives me a working tftp-boot. When it comes to my switch I don't know what I could do, it just relays packages and does not have a webinterface to configure it.

Anyways, after testing my pi's with PXE and NFS I saw them running even slower than me. I will go back to use sd's, maybe I'll try out PiServer in the future. I didn't read about PiServer during my research, why does the official website not mention that piece of software?

And, after this issue this known already, I'll close it.

closed 2018-02-02 by DenSA-Inc