raspberrypi/firmware #935: Can't netboot on Raspberry Pi 1 and 2 with the lastest firmware

Terminus-IMRC, 2018-02-01:
With the latest firmware 58395c1ea3f9ae129885aa7ff7d8b92c7be267db, Raspberry Pi 1 and 2 fail to netboot, but Raspberry Pi 3 does. With the previous firmware 80e1fbeb78f9df06701d28c0ed3a3060a3f557ef, all the models netboot correctly.

tftpd log when netbooting succeeds:
```
Feb  1 12:07:52 imrc dnsmasq-tftp[680]: file /netboot/boot/autoboot.txt not found
Feb  1 12:07:52 imrc dnsmasq-tftp[680]: sent /netboot/boot/config.txt to 192.168.20.103
Feb  1 12:07:52 imrc dnsmasq-tftp[680]: file /netboot/boot/recovery.elf not found
Feb  1 12:07:54 imrc dnsmasq-tftp[680]: sent /netboot/boot/start.elf to 192.168.20.103
Feb  1 12:07:54 imrc dnsmasq-tftp[680]: sent /netboot/boot/fixup.dat to 192.168.20.103
Feb  1 12:07:54 imrc dnsmasq-tftp[680]: file /netboot/boot/recovery.elf not found
Feb  1 12:07:54 imrc dnsmasq-tftp[680]: sent /netboot/boot/config.txt to 192.168.20.103
Feb  1 12:07:54 imrc dnsmasq-tftp[680]: file /netboot/boot/dt-blob.bin not found
Feb  1 12:07:54 imrc dnsmasq-tftp[680]: file /netboot/boot/recovery.elf not found
Feb  1 12:07:54 imrc dnsmasq-tftp[680]: sent /netboot/boot/config.txt to 192.168.20.103
Feb  1 12:07:55 imrc dnsmasq-tftp[680]: file /netboot/boot/bootcfg.txt not found
Feb  1 12:07:55 imrc dnsmasq-tftp[680]: sent /netboot/boot/cmdline.txt to 192.168.20.103
Feb  1 12:07:55 imrc dnsmasq-tftp[680]: file /netboot/boot/armstub.bin not found
Feb  1 12:07:55 imrc dnsmasq-tftp[680]: error 0 Early terminate received from 192.168.20.103
Feb  1 12:07:55 imrc dnsmasq-tftp[680]: failed sending /netboot/boot/kernel.img to 192.168.20.103
Feb  1 12:07:59 imrc dnsmasq-tftp[680]: sent /netboot/boot/kernel.img to 192.168.20.103
Feb  1 12:07:59 imrc dnsmasq-tftp[680]: sent /netboot/boot/bcm2708-rpi-b.dtb to 192.168.20.103
Feb  1 12:07:59 imrc dnsmasq-tftp[680]: sent /netboot/boot/config.txt to 192.168.20.103
```

tftpd log when netbooting fails:
```
Feb  1 12:35:38 imrc dnsmasq-tftp[680]: file /netboot/boot/autoboot.txt not found
Feb  1 12:35:38 imrc dnsmasq-tftp[680]: sent /netboot/boot/config.txt to 192.168.20.103
Feb  1 12:35:38 imrc dnsmasq-tftp[680]: file /netboot/boot/recovery.elf not found
Feb  1 12:35:41 imrc dnsmasq-tftp[680]: sent /netboot/boot/start.elf to 192.168.20.103
Feb  1 12:35:41 imrc dnsmasq-tftp[680]: sent /netboot/boot/fixup.dat to 192.168.20.103
```

When netboot fails, ACT LED flashes 7 times, [which means "kernel.img not found"](https://elinux.org/R-Pi_Troubleshooting#Green_LED_blinks_in_a_specific_pattern). Actually, the pi doesn't load kernel.img after loading start.elf and fixup.dat. So start.elf and/or fixup.dat got something odd between these commits...?

Specifying `start_x=1` or `start_db=1` in `config.txt` doesn't solve the problem.

Could you solve this issue?

pelwell, 2018-02-02:
Today's firmware release contains a patch that should fix this issue.

Terminus-IMRC, 2018-02-03:
Thank you for your so quick response! I really appreciate it! I confirmed it's now fixed, so closing. Thanks!

closed 2018-02-03 by Terminus-IMRC