raspberrypi/firmware #939: MPEG-2 Patents Expired

nsuchy, 2018-02-14:
The patent for MPEG-2 has expired, the firmware should be updated where a license key is no longer required to use the decoder.

**Reference:**
* http://www.mpegla.com/main/programs/M2/Pages/PatentList.aspx
* http://www.mpegla.com/main/programs/M2/Documents/m2-att1.pdf
* http://www.raspberrypi.com/mpeg-2-license-key/

**Note**
Please let me know if this is the wrong repository to bring up this issue.

F5OEO, 2018-02-15:
MPEG-2 encoder should be also very usefull...Surely a firmware update to avoid requiring a licence.

nsuchy, 2018-02-15:
@F5OEO Exactly or at a worst case scenario where Rpis connect to an API which issues the CPU specific license key freebie. I'm not sure how ingrained the need for a license is in the firmware / ROM of whatever chip handles it. I'm sure someone will handle this and it's looking like people agree.

F5OEO, 2018-02-15:
Ping to @popcornmix or @6by9 to inspect implementation , but surely need some feeback from headoffice to remove licencing...Hope that it takes not too long to move on !

nsuchy, 2018-02-15:
@F5OEO Good luck and thanks for looking into this for us :-)

wkeenan, 2018-02-15:
Are the licensing terms of an already existing product (i.e. what Broadcom
baked in) to an end user distinct to being able to implement what an
expired patent describes change anything for the original terms to current
manufactures and their end users?

Are 'we' just hoping the pocket change cost just 'magically evaporate'?

Just asking the question.

Thanks,
Wayne

> Ping to @popcornmix <https://github.com/popcornmix> or @6by9
> <https://github.com/6by9> to inspect implementation , but surely need
> some feeback from headoffice to remove licencing...Hope that it takes not
> too long to move on !

Thanks,
Wayne

nsuchy, 2018-02-15:
The question about the license is if Raspberry Pi Foundation agreed to charge even after the patents expired, on a side note would any court enforce requiring the Raspberry PI Foundation to pay the license fees without a patent, other than for violating a contractual agreement.

6by9, 2018-02-15:
There's no point in pinging me as I'm not involved in the commercial negotiations with MPEG-LA. Without guidance from those who are (probably involving lawyers too), I'm not going to be making any changes.

There are still patents valid in the Philippines and Malaysia, and seeing as Raspberry Pis are sold in those jurisdictions that may still be an issue. There's no simple geographical lock to work around that.

deborah-c, 2018-02-15:
As for the "pocket change" comment, the MPEG-2 license is really 
surprisingly expensive, or at least has been to date: it's a significant 
cost in something with a BOM as small as Raspberry Pi.

WayneKeenan, 2018-02-16:
@deborah-c do you realise that commercial operation systems absorb that cost for you?

WayneKeenan, 2018-02-16:
and actually Im in a bit if a mo fun off with you whiteng,

WayneKeenan, 2018-02-16:
gtf, move on.

deborah-c, 2018-02-16:
I've been involved in video on demand systems since the early trials in the 90s, and video codecs more generally since then, so I'm at least somewhat aware of the economics, but thank you.

However, as with @6by9, I'm not involved in the commercial negotiations, and his comments are very wise

JamesH65, 2018-02-19:
We won't be changing what we do until the final two country patents run out, as stated above, there is no way we can enforce things to the required levels. After that we will look at it again.

closed 2018-02-19 by JamesH65

BryanQuigley, 2020-05-19:
FYI - It's just Malaysia left and they seem to be allowing very old patents to be issued like new (https://bryanquigley.com/pages/mpeg2-patent-tracker.html)