raspberrypi/firmware #942: Keeping a constant i2c clock speed.

ajefr, 2018-02-18:
Hi'
I2C clock speed is derived from cpu_core (BCM2835 peripheral page 34) = cpu_core / CDIV.
For example, if I configure I2C for a 750khz clock, it will work fine in idle mode (cpu_core is 250Mhz)
The problem is when task ask for cpu power, cpu_core will switch to 400Mhz and I2C clock will then switch to ~1.2Mhz.
How could keep a constant i2c clock independant from cpu_core ?
Most i2c peripherals will not accept such high frequencies.
I know that I can force the cpu speed in boot/config.txt but this is a little tricky solution.
Thanks

pelwell, 2018-02-19:
There isn't a way to maintain a constant I2C clock without a level of interlocking between the ARMs and the VPU that we aren't prepared to accept. It would, however, be possible to take an approach similar to the downstream SD driver that delegates clock divisor management to the firmware. Although the resulting clock isn't constant, by carefully ordering core clock & divisor changes we could guarantee that the I2C bus clock is never higher than expected.

The fly in the ointment is that we are currently using the upstream I2C driver, and there is a reluctance to add downstream changes to files that are currently used without modification.

cpagravel, 2018-02-24:
I'd like to add to this. I'm using a Raspberry Pi 3B and I've set the I2C baudrate in config.txt to 160000. If I trigger a few I2C communications, then I read a clock speed of 100kHz on the oscilloscope. However, it seems that if I am performing many more I2C communications, the clock speed changes to 160kHz. I'd like to ensure that the I2C clock speed is constant so I can communicate at the highest possible speed for the devices I'm able to communicate with.

This seems to be related to what ajefr is saying since the clock speed change seems to happen when I'm performing more demanding operations on the Pi.

pelwell, 2018-02-24:
You can work around the issue by adding `core_freq=250` to `config.txt`.

cpagravel, 2018-02-24:
@pelwell Will that have any additional side effects other than keeping the I2C clock speed constant?

pelwell, 2018-02-24:
It will stop the VPU processor clock from going into turbo mode, which can affect the advanced deinterlacer, but it is the same limitation imposed when enable_uart=1 is used on a Bluetooth-equipped Pi. I doubt you will notice a difference unless you are a heavy video or graphics user.

cpagravel, 2018-02-24:
@pelwell I am making an autonomous robot for navigating underwater. I will be localizing using feed from a webcam and performing operations on the images using OpenCV. I'm not sure if OpenCV can use the VPU on the Pi, and I don't know how to check.

pelwell, 2018-02-24:
Be brave and try it.

JamesH65, 2019-11-05:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2019-11-05 by JamesH65

ajefr, 2019-12-31:
With the new RPI4, we now have another default core frequency (500Mhz)
How could I, within a kernel module, know at wich speed runs the core ?
It's easy within userspace with vcgencmd but I haven't found an easy way in kernel space.

pelwell, 2020-01-06:
The Pi DTBs configure the I2C controllers, including a reference to their master clock:
```
	clocks = <&clocks BCM2835_CLOCK_VPU>;
```
If your driver/module uses DT it can use a similar configuration to configure its clock, then call `clk_get_rate` on that clock. The clock should also be available as the named clock "vpu".

Alternatively you can use the firmware interface to send the GET_CLOCK_RATE mailbox message.

kripton, 2020-04-10:
> You can work around the issue by adding `core_freq=250` to `config.txt`.

@pelwell : Would you mind elaborating how setting a value to it's default value changes the behaviour? Is some "automatic clock adjustment" disabled when the value is set explicitely?

Thanks :)

pelwell, 2020-04-10:
Because it's not the default value. Don't confuse the setting `core_freq` with the actual core frequency - it specifies the maximum/turbo frequency for the VPU/core clock. The "default" value, as much as there is such a thing, is set by `core_freq_min`. I'm sure you can see how setting the minimum and the maximum for something may stop it changing.

Be aware that 250MHz is not guaranteed to be the minimum/non-turbo clock on all Pis, but it is a safe value. At some point we may get round to adding a way to force just the core clock to be at its turbo or non-turbo speed without changing, but until then setting core_freq and core_freq_min to the same value is the way to do so.

kripton, 2020-04-10:
Thanks for the explanation, that makes sense now. On https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md it reads a bit different to me: `... default value is 250 for Pi 1/Pi 2 ...`
True, setting the minimum and maximum values should keep it fixed to that clock. I'll experiment with that a little.

pelwell, 2020-04-10:
You can always use `vcgencmd get_config core_freq` (and core_freq_min) to find out what the default values are before you change them.

kripton, 2020-04-13:
Hi @pelwell : I've tried setting up something that should be reproducible to resolve my problem with Jitters on DMA -> GPIO transfers (which I still think might be related to unsteady frequency of clocks).
You can read about what I tried and what the results are here: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269035&p=1642694#p1642532
The video mentioned are linked in the post before the linked one.

Main takeaways: The clock settings in config.txt do actually affect the timing behaviour of the GPIOs (at least when using pigpios' WaveChain functions). However, I was unable to find a combination that gets me rid of the jittering completely.

pelwell, 2020-04-13:
What does that have to do with the original issue?

kripton, 2020-04-13:
Well, it's both about the timing on the GPIOs being affected by some clock adjustment, depending on load. I agree that my use case is completely different. However, my research has brought me here and the comments here about being able to set the clocks to fixed values affected what I am seeing as well. Prefer discussing it in a new issue report instead?

pelwell, 2020-04-14:
> Prefer discussing it in a new issue report instead?

Yes - one problem per "issue", please.