raspberrypi/firmware #945: Official 7 ich touch screen: Touch out of calibration after firmware update.

fosspill, 2018-02-20:
I believe this is a firmware issue since the only cause for this issue is updating the firmware.

The touch works perfectly at the top of the display, but the further down you touch you'll have more offset between your finger and the cursor.
All the way to the bottom of the screen the offset is about 1cm.

I've tried with and without overscan, no change. Only thing that helps is to go back to the old firmware.

JamesH65, 2018-02-20:
I think I have found a problem with the device tree overlay - I accidentally set the vertical resolution to 600, not 480. Doh. Anyway, pushing a fix though to 4.9 and 4.14, in the mean time you can manually change the overlay, it's called rpi-ft5406-overlay.dts, edit it to replace the 600 with 480 and it should now scale correctly.

pelwell, 2018-02-20:
Compile the modified overlay with:
```
$ dtc -@ -I dts -O dtb -o rpi-ft5406.dtbo rpi-ft5406-overlay.dts
```
Copy the resulting .dtbo to /boot/overlays.

fosspill, 2018-02-20:
pi@raspberrypi:~ $ uname -r
4.14.18-v7+

closed 2018-02-20 by fosspill

JamesH65, 2018-02-20:
That's great - thanks for reporting it.