raspberrypi/firmware #948: `config.txt` has no effect in raspberry pi 2

WeiqiChenCN, 2018-02-24:
I manually create mbr partitions, format partitions and download `firmware/boot` files(latest version), also I deploy rootfs.

Details:

content of `config.txt` ( I manually created it):
```bash

$ cat /boot/firmware/config.txt
cmdline=dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=UUID="239f429f-e438-4a52-b756-0e6a5e0973c7" rootfstype=ext4 root_wait
kernel=kernel7.img

```

Now, I did successfully boot in raspberry pi 2, but **the file `config.txt` has no effect at all, In fact it does not change the cmdline**:

```bash
$ cat /proc/cmdline
bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
```

Other information: 

```bash

$ sudo fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 14.9 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x9b950365

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        2048   526335   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      526336 31116287 30589952 14.6G 83 Linux

$ blkid
/dev/mmcblk0p1: SEC_TYPE="msdos" UUID="8954-5DC9" TYPE="vfat" PARTUUID="9b950365-01"
/dev/mmcblk0p2: UUID="239f429f-e438-4a52-b756-0e6a5e0973c7" TYPE="ext4" PARTUUID="9b950365-02"

$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 14.9G  0 disk
├─mmcblk0p1 179:1    0  256M  0 part /boot/firmware
└─mmcblk0p2 179:2    0 14.6G  0 part /

$ ls /boot/firmware/
bcm2708-rpi-0-w.dtb     config.txt     LICENCE.broadcom
bcm2708-rpi-b.dtb       COPYING.linux  start_cd.elf
bcm2708-rpi-b-plus.dtb  fixup_cd.dat   start_db.elf
bcm2708-rpi-cm.dtb      fixup.dat      start.elf
bcm2709-rpi-2-b.dtb     fixup_db.dat   start_x.elf
bcm2710-rpi-3-b.dtb     fixup_x.dat    System Volume Information
bcm2710-rpi-cm3.dtb     kernel7.img
bootcode.bin            kernel.img

$ cat /etc/fstab
# UNCONFIGURED FSTAB FOR BASE SYSTEM
UUID=239f429f-e438-4a52-b756-0e6a5e0973c7 / auto defaults 0 0
UUID=8954-5DC9 /boot/firmware auto defaults 0 0

```

pelwell, 2018-02-24:
You have put the configuration file in the wrong place. The correct path from the root of the FAT32 partition of the SD card is `/config.txt`, which gets mounted as `/boot/config.txt`.

WeiqiChenCN, 2018-02-24:
@pelwell Yeah,
I did modify `config.txt` in FAT32 partition.
Refer:

```bash
$ cat /etc/fstab
# UNCONFIGURED FSTAB FOR BASE SYSTEM
UUID=239f429f-e438-4a52-b756-0e6a5e0973c7 / auto defaults 0 0
UUID=8954-5DC9 /boot/firmware auto defaults 0 0

$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 14.9G  0 disk
├─mmcblk0p1 179:1    0  256M  0 part /boot/firmware
└─mmcblk0p2 179:2    0 14.6G  0 part /
```

I mount the FAT32 partition at `/boot/firmware`.

WeiqiChenCN, 2018-02-24:
I am reminded of that --  UUID ONLY WORKS when you have an initramfs. I should use 

```
cmdline=root=PARTUUID="9b950365-02" rootfstype=ext4
```
or
```
cmdline=root=/dev/mmcblk0p2 rootfstype=ext4 
```

pelwell, 2018-02-24:
It is not clear how your cmdline parameter is split across lines (out shouldn't be), but I can tell you that there is a limit to how long lines may be - it was 80 characters but it was increased, perhaps to 100 or 120.

I have never used the cmdline= setting, and I don't have a Pi to hand, so let's assume for a moment that it is broken in some way or at least not behaving the way you expect it to. Try putting your kernel parameters in cmdline.txt instead.

WeiqiChenCN, 2018-02-24:
I download Ubuntu Mate for Raspberry Pi image (took some time) from [here](https://ubuntu-mate.org/raspberry-pi/ubuntu-mate-16.04.2-desktop-armhf-raspberry-pi.img.xz), then open it with Diskgenius( on Windows ), extract `config.txt` and `cmdline.txt`, and put them into my sd card.

I did some modification , it works now with `cmdline.txt`:

```shell
$ cat /proc/cmdline
bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
$ cat /boot/firmware/cmdline.txt
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

and according to the content of `config.txt`,
```
## cmdline (string)
##     Command line parameters. Can be used instead of cmdline.txt file
##
#cmdline=""
```

I still don't understand why `cmdline` not work, Is it possible it failure because i did not use `""` before???

pelwell, 2018-02-24:
I don't know yet, but it shouldn't be hard to figure out when I have time.

WeiqiChenCN, 2018-02-24:
I tried `sudo mv config.txt config.txt.bak`, then reboot. `cmdline.txt` still works! 
OK, I think your are right, put it in cmdline.txt likely is a good idea.

pelwell, 2018-02-24:
I don't know where your config.txt came from - it looks a bit like the standard one, but it clearly isn't -
but the explanation of `cmdline` is wrong. It is used to specify an alternative filename from which to read the kernel command line. See the official documentation - unsurprisingly it is correct: https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md

WeiqiChenCN, 2018-02-25:
I created config.txt manually before..

closed 2018-02-25 by WeiqiChenCN