raspberrypi/firmware #950: Rotate HDMI display does not work in raspberrypi

shashankpr, 2018-03-10:
I have a Raspberry Pi3 model (with Raspbian stretch) and I have connected it to my monitor via HDMI (HDMI-HDMI). 
I have checked the online forums and performed the following steps:
- sudo rpi-update (now I have the latest firmware)
- sudo apt-get update && apt-get upgrade
- Set ***force_hdmi_plug=1***, **_hdmi_group=2_**, **_hdmi_mode=2_**
- Set ***gpu-mem=256***
- Set ***display_hdmi_rotate=1*** at the end of /boot/config.txt

I have checked almost every forum and help and yet after rebooting my Pi, the screen does not change its orientation.

Please guide me or let me know where I am going wrong.

JamesH65, 2018-03-10:
Hmm, now you are making me wonder whether that option has been included in the code yet. I'll need to check on Monday. I thought it had been merged, but maybe not.

In the meantime, you could try just display_rotate=1.

pelwell, 2018-03-10:
It could be a typo in your post rather than your config file, but `gpu_mem=256` should have an underscore, not a hyphen.

shashankpr, 2018-03-10:
@JamesH65 
Initially I tried with _display_rotate=1_ 
After checking the RaspberryPi _config_ doc (it mentioned that _display_rotate_ is deprecated), I changed it to _display_hdmi_rotate_=1

JamesH65, 2018-03-10:
So did display_rotate not work? The changes I made (if they are merged)  should be backward compatible.

Ah, had a thought - what resolution display? IIRC, there was an odd limitation of rotate that meant anything over a particular width was ignored. I removed that for the hdmi rotate stuff.

shashankpr, 2018-03-10:
This is the output of my `vcgencmd get_config int` : 

arm_freq=1200
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=400
desired_osc_freq=0x36ee80
disable_commandline_tags=2
disable_l2cache=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
display_rotate=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=300
hdmi_drive=2
hdmi_force_cec_address=65535
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=81
init_uart_clock=0x2dc6c00
lcd_framerate=60
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=26370
over_voltage_avs=0x1cfde
pause_burst_frames=1
program_serial_random=1
sdram_freq=450

shashankpr, 2018-03-10:
@JamesH65 
No, `display_rotate=1` didn't work either.
The resolution I had set is 1366*768 which is the resolution of display monitor.

JamesH65, 2018-03-10:
Ah, OK, then the option is in there. -1 is the default if it is unset on config.txt. Not sure why it isn't working then.

JamesH65, 2018-03-10:
Did you check the post from @pelwell  above? Could be running out of VC memory, it's quite a big rotate.

shashankpr, 2018-03-10:
I tried the same methods with 2 different monitors, one connected directly via HDMI and the other though HDMI-VGA adapter, but the neither of the displays rotated.

Should I try setting a higher `gpu_mem` value ?

JamesH65, 2018-03-10:
256 should be more than enough.

Cannot really check anything else until Monday,

shashankpr, 2018-03-10:
@JamesH65 No worries! I just got a bit frustrated after having tried every possible thing and decided to post it here.
@pelwell : Yeah, sorry! Its a typo in my post. The config file has it with an underscore.

6by9, 2018-03-10:
@shashankpr You have 
```
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=26370
```
That would imply to me that you're using the GL and KMS driver, not the firmware drivers. In which case none of the config.txt controls for hdmi and display controls will work as it is all down to the Linux kernel drivers and not the firmware.
You need to use the legacy driver (or Fake KMS?) if you want to use the config.txt settings, otherwise it's xrandr to set things up.

mwoolweaver, 2018-03-10:
This is my config.txt that is working on my Pi3 

gpu_mem=128

display_rotate=1 

[config.txt](https://github.com/raspberrypi/firmware/files/1799550/config.txt)

shashankpr, 2018-03-13:
@6by9 

Thanks a lot. Your suggestion worked. I switched back to legacy drivers and used `display_rotate=1` and it works now. I am going to close the issue now.

closed 2018-03-13 by shashankpr

tungmv7, 2019-04-27:
> @shashankpr You have
>
> ```
> mask_gpu_interrupt0=1024
> mask_gpu_interrupt1=26370
> ```
> That would imply to me that you're using the GL and KMS driver, not the firmware drivers. In which case none of the [...]

thanks :) you saved my life