raspberrypi/firmware #951: Default UART clock rate seems to be too high (1 GHz)

rsta2, 2018-03-16:
With the firmware 086a848 from Mar 14 the default UART clock rate is 1 GHz. Is this intended? It seems to be to high, because with the IBRD baud rate divisor set to 542, which should result in about 115200 Bps, I get no serial output. When setting the UART clock rate to 48 MHz using the "Set Clock Rate" property mailbox call the serial output does work successfully. This has been tested on a Raspberry Pi 2B v1.1 with a bare metal test program.

pelwell, 2018-03-16:
Assuming you've not updated for a while, this firmware release includes a number of internal reorganisations (including clock calculation) prompted by changes required for 3B+, some of them debuting with that release. so a bug in that area isn't impossible. Having said that, it's not one I've seen.

Removing references to the serial console from cmdline.txt and adding "dtparam=uart0=off" to config.txt, I get the expected UART clock of 48MHz (and no Linux driver to use it...). Looking at the code I can see one reference to 1GHz in relation to the UART clock, but I can't see a scenario where it would be used.

Can you post the content of your config.txt?

pelwell, 2018-03-16:
Please also outline the steps your test program goes through that could be relevant - primarily mailbox calls and hardware accesses.

rsta2, 2018-03-16:
I do not use the config.txt. Because this is a bare metal environment the only important files on the SD card are bootcode.bin, start.elf, fixup.dat and the kernel7.img of the test program. I think this is a common environment for bare metal programming, where device tree files are not used.

My test program does basically the following things:

1. Leaves HYP mode and initialises the stack pointers
2. Reads the board revision code via mailbox
3. Reads the size of the ARM memory via mailbox and initialises the heap
4. Initialises the frame buffer using these mailbox tags in a row:
	SET_PHYS_WIDTH_HEIGHT
	SET_VIRT_WIDTH_HEIGHT
	SET_DEPTH
	SET_VIRTUAL_OFFSET
	ALLOCATE_BUFFER
	GET_PITCH
5. Reads the UART0 (PL011) clock using the mailbox call GET_CLOCK_RATE (0x30002, clock ID 2)
6. Calculates the baud rate divisors (Int 542, Frac 34) and initialises the UART0 with it
7. Writes to the UART (the written data cannot be received by an external terminal program)

I could append the test program here as binary (kernel7.img) and source code, if this helps. It uses an external bare metal library, which is available on GitHub.

pelwell, 2018-03-16:
> 5. Reads the UART0 (PL011) clock using the mailbox call GET_CLOCK_RATE (0x30002, clock ID 2)

Does this return 1GHz or 48MHz?

pelwell, 2018-03-16:
I'm going to assume that GET_CLOCK_RATE is returning 1GHz, which leads to the wrong divisor being calculated because a) the real clock is actually 48MHz and b) the divisor probably doesn't have enough range to get you down to 115200.

@popcornmix has a cunning plan.

rsta2, 2018-03-16:
> Does this return 1GHz or 48MHz?

Yes, it returns 1 GHz. I think, the calculation of the divisor is right. The integer part has 16 bits, so this should be enough.

I did another test now. I manually set the clock rate as input to the calculation to 48 MHz. That is working. So that means, only the reported value of 1 GHz is wrong. The real clock is actually 48 MHz, as you expected.

What plan?

popcornmix, 2018-03-16:
@rsta2 I wonder if you could test this firmware?
https://drive.google.com/uc?id=1YkQrqAHeq6suhZHxMIBwedf7ZgrFeDl2&export=download

rsta2, 2018-03-16:
@popcornmix With this firmware everything is all right. Reports 48 MHz and the serial output arrives at the terminal.

popcornmix, 2018-03-16:
Thanks for testing. Fix should be present in latest rpi-update firmware.

rsta2, 2018-03-16:
I tested the firmware 25cf637 from the repo again. It is working well. Thanks.

closed 2018-03-16 by rsta2