raspberrypi/firmware #953: issues with Samsung EVO card on Mar 7th firmware

nullr0ute, 2018-03-16:
In Fedora we've had a few reports of rainbow screens and not booting and it took a little to track down the issue. We usually see the rainbow colours with PSU problems.

The details took a little to work out but it seems with the March 7th (7fdcd00e00a42a1c91e8bd6f5eb8352fe9358557) and later firmwares when used on a Samsung EVO (not sure if there's a particular subset, or if it affects other cards) micro SD give us the rainbow output and doesn't boot further. Other cards such as Sandisk Ultra work fine.

popcornmix, 2018-03-16:
Can you confirm that March 7th firmware has this issue, and previous March 5th firmware does not have this problem?

paulwhalen, 2018-03-16:
The March 5th firmware worked as expected.

Using serial console it hangs here:

U-Boot 2018.03-rc4 (Mar 08 2018 - 20:42:49 +0000)

DRAM:  998 MiB
RPI 3 Model B (0xa22082)
MMC:   mmc@7e202000: 0, sdhci@7e300000: 1
Loading Environment from FAT...

popcornmix, 2018-03-16:
To help narrow down the issue can you confirm if this test firmware is good?
https://drive.google.com/uc?id=1pwZSRw4BuoWYixZHKuhE42lypraa-eG6&export=download

nullr0ute, 2018-03-16:
Correct, March 7th has the issue, March 5th does not.

pelwell, 2018-03-16:
A recent change set the EMMC clock to 200MHz in order to allow the SDIO interface to run at 50MHz (rather than 41.6MHz with a source clock of 250MHz - the EMMC divisor must be even), and I think this is the cause of the problem.

@agraf There's a bug in the bcm2835_sdhost port to U-boot, specifically [this line](https://github.com/u-boot/u-boot/blob/master/drivers/mmc/bcm2835_sdhost.c#L944):
```
host->max_clk = bcm2835_get_mmc_clock();
```
Although that looks reasonable, the SDHost interface actually shares the core/VPU clock, so you'll need another getter function (or a parameterised version if bcm2835_get_mmc_clock, or a generic bcm2835_get_clock that takes a clock number, etc.) that queries the core clock instead.

paulwhalen, 2018-03-16:
popcornmix, the test fw worked ( https://drive.google.com/uc?id=1pwZSRw4BuoWYixZHKuhE42lypraa-eG6&export=download )

popcornmix, 2018-03-16:
Okay, that confirms your problem is related to the EMMC clock change (that was the only part reverted), but I think the fix for this will have to come from uboot.

pelwell, 2018-03-16:
Is it also worth trying this afternoon's regular firmware release, just to be sure it wasn't related to the CLOCK_GET_RATE issue?

jonathangray, 2018-03-17:
This also occurs with a lexar uSD card with rpi3 here.  Any version of start.elf after aa755336fc3f0430fb9175c9fe5f0e1d27f7dd54 hangs at 'Loading Environment from FAT...' when using U-Boot.  This includes 
25cf637ccc90d7d2fa37277c807ab33b655bd0f4 which mentioned the CLOCK_GET_RATE change in the commit message.

U-Boot> mmc info 
Device: mmc@7e202000
Manufacturer ID: 28
OEM: 4245
Name:       
Bus Speed: 50000000
Mode : SD High Speed (50MHz)
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 14.9 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes

jonathangray, 2018-03-17:
Changing U-Boot to fetch the frequency for SDhost using the core clock id as suggested by @pelwell allows me to use U-Boot with the 25cf637ccc90d7d2fa37277c807ab33b655bd0f4 firmware.

https://patchwork.ozlabs.org/patch/887255/

jonathangray, 2018-05-08:
U-Boot 2018.05 includes the patch (8ae1f8298820a19380bafc07ad554bd629cc27be).  This issue can be closed.

closed 2018-05-08 by pelwell