raspberrypi/firmware #954: where is the firmware for 4.9.80+/-v7+ #1098

MrEngman, 2018-03-16:
would be helpful if hexxeh/rpi-firmware and raspberrypi/firmware were updated so I can compile the drivers I do for those kernel versions.

macmpi, 2018-03-22:
you may check https://github.com/Hexxeh/rpi-firmware/issues/174

popcornmix, 2018-03-22:
stable firmware tree should now include 4.9.80 kernel and firmware from latest raspbian sdcard image.

MrEngman, 2018-03-22:
And git_hash now points to a kernel commit, `921b625445`, that doesn't appear to exist.

popcornmix, 2018-03-22:
That is hard to avoid. The raspian kernel was built before the Pi3+ was announced, and so the changes hadn't been pushed publicly.
After announcement the changes were pushed to linux tree, but don't have the same hash (perhaps commit order is different).

MrEngman, 2018-03-22:
Not very helpful. However, care to suggest which commit I could use then, if it's available.

Maybe `d584633` ??

MrEngman, 2018-03-22:
Hi popcornmix,

Sorry to keep nagging on about issues with the firmware repos but the latest commit for 4.9.80 has another problem. Just ran `sudo rpi-update 936a8dc3a605c20058fbb23672d6b47bca77b0d5` to install 4.9.80 #1098 on one of my Pis and it came up with the following:-

```
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
This update bumps to rpi-4.9.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    252      0 --:--:-- --:--:-- --:--:--   253
100 87.4M  100 87.4M    0     0   223k      0  0:06:41  0:06:41 --:--:--  146k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.80+
 *** depmod 4.9.80-v7+
 *** depmod 4.9.59-v7+
 *** depmod 4.9.59+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 936a8dc3a605c20058fbb23672d6b47bca77b0d5
 *** A reboot is needed to activate the new firmware
```

taudac, 2018-03-23:
@popcornmix , @MrEngman can you point me to the corresponding kernel commit?

popcornmix, 2018-03-23:
I've pushed the tree used to build last raspbian kernel:
https://github.com/raspberrypi/linux/commits/rpi-4.9.y-stable
The top commit is the one from git_hash: https://github.com/raspberrypi/linux/commit/921b6254452db87ae0304beaa9833cfdf5083ed8

Also removed the 4.9.59 kernel modules from stable firmware repo.

popcornmix, 2018-04-03:
@MrEngman is this issue resolved now?

MrEngman, 2018-04-03:
yes, closing.

closed 2018-04-03 by MrEngman